dresco/STM32H7xx #1: SKR3 EZ - USB port not working.

Demitrius, 2022-09-06:
Hello Jon!

I build today latest version. Board flashed thru SD card with new firmware.

Windows not found device (board) when I connect board USB to PC.

PS: I setup TMC 2209 (X axis) and TMC 5160 EZ Pro (Y axis). Marlin 2 can drive both motors.

dresco, 2022-09-06:
Hi, I suspect it's related to using the Marlin bootloader. If you are happy to flash over USB (which will overwrite the Marlin SD card bootloader), then I would try that first?

Entering the USB bootloader mode should just be question of holding down the boot button while you press & release reset, and then release the boot button. You should see the DFU device appear instead of USB serial.. Must admit I don't know whether that needs a driver on Windows.

btw, I'm not sure whether having different TMC drivers on different axis is supported in grblHAL? I know I've never tested it, probably a question for Terje..

Demitrius, 2022-09-06:
I attach board over USB in DFU mode and connect thru STM Cube Programmer. My next steps?
[image]

Demitrius, 2022-09-06:
I read 128K of current bootloader and save to file, now I need upload other bootloader?

dresco, 2022-09-06:
You can flash the grblHAL firmware file directly, as the DFU bootloader is hard coded into these chips.

The STM2CubeProgrammer can program the elf file directly. Once built - it will be in your STM32CubeIDE project workspace, either `Debug BTT SKR 3/grblHAL Driver STM32H7xx.elf` or `Release BTT SKR 3/grblHAL Driver STM32H7xx.elf`

The 2nd green icon on the left is "Erasing and Programming", click on that, browse to the firmware file, & start programming..

[image]

Demitrius, 2022-09-07:
Hi

Flashed and now USB works. I do not found Debug version of .elf, only release firmware.elf file avail.

I connect TMC2209 driver and on stepper motor (X slot).
Trying to use Universal Gcode Platform to connect to GRBL and move single motor (X) but have ALARM message.

Something not enought?
Hi

Flashed and now USB works. I do not found Debug version of .elf, only release firmware.elf file avail.

I connect TMC2209 driver and on stepper motor (X slot).
Trying to use Universal Gcode Platform to connect to GRBL and move single motor (X) but have ALARM message.

Something not enought?

[image]

dresco, 2022-09-07:
That's progress :)

I would maybe connect from a  terminal client like PuTTY to check the cause of the alarm. The real time report command `?` should help there. Quite often is just something simple like the expected normally closed inputs not being wired up..

[This](https://github.com/grblHAL/Plugins_motor) page is useful for the commands to interact with the TMC drivers (M122 etc). If it is driver problems, then can be useful to change the axis mask ($338) to enable just the single X axis for testing.

dresco, 2022-09-08:
> Have you idea what is wrong?

Yep, error 18 is because it thinks e-stop is pressed.. Have a look at the [first run settings](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) page. You'll need to invert the logic for the missing inputs - as they are expected to be normally closed. Try `$14=7` for starters

Demitrius, 2022-09-08:
Yes, this works. I can drive motors (X100, Y100 commands)

Jon, question. Wifi module attached, can I use it? Also, how to use SD card?

Demitrius, 2022-09-08:
Yes, I want inspect this part of project. Networking (over WiFi) will very useful.

Jon, does it enough enable PPI_ENABLE to use RGBL in laser mode?

dresco, 2022-09-08:
> What do you mean?

I was assuming you wanted to test the laser plugin by defining PPI_ENABLE and rebuilding? 

Fwiw - I've never built that plugin, and don't know whether it's even been tested in the upstream F7 driver (that the H7 driver is based on) - so you are quite probably on the bleeding edge! ;)

Demitrius, 2022-09-08:
Read networking and WiFi discussion. As I understand, WebUI have dependency with grbl and network. Need to split logic before use in different configurations.

Demitrius, 2022-09-08:
Jon, will we test TMC5160 drivers?

dresco, 2022-09-08:
> Jon, will we test TMC5160 drivers?

Yes sure. I don't have the EZ modules to test with my SKR 3, but I've used a 5160 breakout with my dev board, so I think it should work fine - as long as all the drivers are the same type - I don't think you can mix & match axis.. 

If you open Properties from the Project menu, then select either the Debug or Release build for the SKR3. Then following this screenshot, you'll want to change the TRINAMIC_ENABLE value to 5160 and rebuild/flash.

[image]

Demitrius, 2022-09-08:
Works!

[image]

Demitrius, 2022-09-08:
Jon, how about speed ?
I see some videos with same motors/steppers and it looks a more faster.

dresco, 2022-09-08:
> Jon, how about speed ?
> I see some videos with same motors/steppers and it looks a more faster.

Probably just default (conservative) max speed and acceleration settings? You can list all the settings with `$$`

On a bare board for testing, you can crank them up no problem, this is what I normally set mine to for bench tests. (Obviously when you're building a machine, then you need to work up and see what's achievable / reliable).

```
$110=3500	X-axis max rate (mm/min)
$111=3500	Y-axis max rate (mm/min)
$112=3500	Z-axis max rate (mm/min)

$120=300.0	X-axis accel (mm/sec^2)
$121=300.0	Y-axis accel (mm/sec^2)
$122=300.0	Z-axis accel (mm/sec^2)
```

Demitrius, 2022-09-08:
You are right! Set 3500 and looks better.

Demitrius, 2022-09-08:
Now I need solve laser question and can start build laser cutter

Main question about commands, as I see, some peoples use commands like M3, M4 to set laser static/dynamic mode. How about other like laser power? (thru PWM?)

dresco, 2022-09-09:
> Now I need solve laser question and can start build laser cutter

I think these are questions for the other thread, are you okay to close this issue now the board is up & running?

Demitrius, 2022-09-09:
Yes Jon, you did a great job!
Thank you for your attention and help with GRLB, SKR3EZ and drivers.

closed 2022-09-09 by Demitrius

Avalonnw, 2023-01-21:
Sorry for jumping in to the old closed thread.
I'm moving from grbl-mega and struggling. Nothing I do seem to move any motors for some reason. They don't even feel energized. I'm using SKR 3 EZ with 5160EZ drivers, powering all from 24v supply. I've inverted limit switches and all safety signals, ioSender allows me to jog, but no movement happens.
I'm not that stupid usually, I promise...

dresco, 2023-01-21:
Off the top of my head, I would check you're compiling with `TRINAMIC_ENABLE` set to `5160`, and that you have the `$338` axis mask set correctly..

Might be further useful info on [this](https://github.com/grblHAL/Plugins_motor) page. Does the M122 debug output from the drivers look sane?

Avalonnw, 2023-01-22:
M122
[MSG:Warning: Could not communicate with stepper driver!]

Maybe that's why...

Avalonnw, 2023-01-22:
Victory!
Had to re-pull the code and re-configure it.
Web builder does not activate the comms with TMC drivers. Also could not configure in CUBE, so had to run it in Platformio.

dresco, 2023-01-22:
> Victory! Had to re-pull the code and re-configure it. Web builder does not activate the comms with TMC drivers. Also could not configure in CUBE, so had to run it in Platformio.

Thanks for the update, I'll check the Web Builder configuration.

STMCubeIDE should work though, what happens there? Worth noting that I set `TRINAMIC_ENABLE=2209` inside the project properties for the SKR board, so would need to be changed to 5160 there..
[image]

Avalonnw, 2023-01-22:
Changed to 5160 in cube and there was no difference for some reason.