dresco/STM32H7xx #5: Hello, what is the maximum output frequency of the h743 pulse?

hanke-cnc, 2022-11-05:
Sorry, I have sold my h7 development board before, there is no way to test, I have tested the f401, the stable output is 300khz, the f446 is 400khz

dresco, 2022-11-06:
Hi, I can test for you. How are you comparing, just a single axis move, or while running a more typical job? Cheers..

hanke-cnc, 2022-11-06:
> Hi, I can test for you. How are you comparing, just a single axis move, or while running a more typical job? Cheers..

Thank you so much!! Only one axis movement

dresco, 2022-11-07:
Will get to 700khz stable output on a single axis move, is glitchy by 800khz..

hanke-cnc, 2022-11-07:
> Will get to 700khz stable output on a single axis move, is glitchy by 800khz..

Danke, so cool!!

hanke-cnc, 2022-11-07:
> Will get to 700khz stable output on a single axis move, is glitchy by 800khz..

hi, dresco, I haven't tried servo motors, I heard that servo motors need high pulse output control?

closed 2022-11-07 by hanke-cnc