dresco/STM32H7xx #6: Commits for SKR3 and Dual spindle option

JoachimF, 2022-11-25:
Hello Dresco,

Here is some corrections for the SKR3 configuration version :

### remove in the preprocessors directives : 
 - TRINAMIC_ENABLE=2209
 - SDCARD_ENABLE=1
 - 
### Code correction : 
in driver.c
line 445
  '#ifdef Y2_ENABLE_PORT
   ' DIGITAL_OUT(Z2_ENABLE_PORT, Z2_ENABLE_BIT, enable.z;)
  '#endif
to
  '#'ifdef **Z2_ENABLE**_PORT
   'DIGITAL_OUT(Z2_ENABLE_PORT, Z2_ENABLE_BIT, enable.z' **);**
  '#endif

line 437
'  #ifdef X2_ENABLE_PORT
'    DIGITAL_OUT(X2_ENABLE_PORT, X2_ENABLE_BIT, enable.x;)
'  #endif
'    DIGITAL_OUT(Y_ENABLE_PORT, Y_ENABLE_BIT, enable.y);
'  #ifdef Y2_ENABLE_PORT
'    DIGITAL_OUT(Y2_ENABLE_PORT, Y2_ENABLE_BIT, enable.y;)
'  #endif
to
'  #ifdef X2_ENABLE_PORT
'    DIGITAL_OUT(X2_ENABLE_PORT, X2_ENABLE_BIT, enable.x);
'  #endif
'    DIGITAL_OUT(Y_ENABLE_PORT, Y_ENABLE_BIT, enable.y);
'  #ifdef Y2_ENABLE_PORT
'    DIGITAL_OUT(Y2_ENABLE_PORT, Y2_ENABLE_BIT, enable.y);
'  #endif

Also, I enabled the dual spindle function, but M104P0 is not reconized, and when  I tested the VFD with M3S0, the processor reset.

Thanks for your help

Joachim

dresco, 2022-11-26:
Hi @JoachimF 

Thanks for the code corrections, will push a fix this afternoon. (I'll also raise a PR for the upstream [F7](https://github.com/grblHAL/STM32F7xx) driver, as were inherited from there).

I'll think a bit more about the preprocessor directives. Including them in the IDE settings is useful to me, as I'm always switching between boards for testing, so I don't make any changes to my_machine.h. I guess that is the problem for you, that the IDE settings get in the way of what you want to define locally in my_machine?

I've got a Huanyang VFD on the bench here, so will test the spindle control..

Regards,
Jon

JoachimF, 2022-11-26:
I understand for the preprocessor directives, as I change mcu targets and IDE, I always spend time to find things, and I thought I was silly when trinamic was enabled whereas the option was not enabled and avoid me to enable VFD

I have an ESP32 target and I have the same problem with the spindle, it crash the cpu on command send, maybe a plugin problem.

I have 2 CNC that need dual spindle (VFD and Hotwire/Laser), ESP32 also don't have this option

Regards,

Joachim

dresco, 2022-11-26:
Seems to be working for me (on my dev board at least, will dig out the SKR3 later and test on that also).

What sort of VFD are you using - am assuming you have set your VFD type with the appropriate `VFD_ENABLE` define? https://github.com/grblHAL/Plugins_spindle/blob/6b70b5cfe2f3573b0cb33bd178e08c4e29b76353/shared.h#L32-L38

The multiple spindle support looks okay, but is not obvious to configure, and I'm not convinced `DUAL_SPINDLE` actually does anything..

What I had to do was increase `N_SPINDLES` from  the default value of 1, in order to get a `[SPINDLE:xxx]` entry from `$I`. `$$=395` should then list the available spindle ids, which can then be selected at runtime with `M104Qn` where n is the wanted spindle id.

dresco, 2022-11-26:
> I thought I was silly when trinamic was enabled whereas the option was not enabled and avoid me to enable VFD

Btw, I think I can safely remove the compilation check that prevents enabling modbus when trinamic uart (2209) is also enabled.. The intent is to stop multiple plugins trying to use the same serial port, but the trinamic code in the H7 driver uses a soft uart anyway.

dresco, 2022-11-26:
Hi @JoachimF

Can confirm the VFD support is working for me here on my SKR3 board. Am taking 3.3v and ground from the wireless header, and PA9/PA10 from the TFT header;
[image]

For RS485, am just using the cheap transceivers from AliExpress, which seem to work fine at 3.3v;
[image]

```
$I
[VER:1.1f.20221101:]
[OPT:VNMSL,1024,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,TC,SED,TMC=7,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32H743]
[DRIVER VERSION:221014]
[BOARD:BTT SKR-3]
[PLUGIN:Trinamic v0.09]
[PLUGIN:MODBUS v0.14]
[PLUGIN:HUANYANG v1 VFD v0.08]
[SPINDLE:Huanyang v1]
[PLUGIN:SDCARD v1.07]
ok
```

```
$PINS
[PIN:PA4,Reset]
[PIN:PA5,Feed hold]
[PIN:PA6,Cycle start]
[PIN:PC13,Probe]
[PIN:PC1,X limit min]
[PIN:PC3,Y limit min]
[PIN:PC0,Z limit min]
[PIN:PD4,X step]
[PIN:PA15,Y step]
[PIN:PE2,Z step]
[PIN:PD3,X dir]
[PIN:PA8,Y dir]
[PIN:PE3,Z dir]
[PIN:PD6,X enable]
[PIN:PD1,Y enable]
[PIN:PE0,Z enable]
[PIN:PD5,UART X]
[PIN:PD0,UART Y]
[PIN:PE1,UART Z]
[PIN:PB6,Spindle on]
[PIN:PB5,Spindle direction]
[PIN:PB3,Flood]
[PIN:PB4,Mist]
[PIN:PA10,RX,Modbus]
[PIN:PA9,TX,Modbus]
[PIN:PB0,Spindle PWM]
ok
```

JoachimF, 2022-11-27:
Hello Jon,

I see that I have the same RS485 converter.
I made progress on the ESP32 target, the STM is not at home. I find how to use dual spindle, it's in the config.h of GRBL

'#ifndef N_SPINDLE
'/*! Defines number of spindles supported - minimum 1, maximum 32
'#define N_SPINDLE 2
'#endif

Now, I can run the VFD, but the swap between crash the mcu, on the ESP32, I haven't tested on the STM yet.
I'll give you the result soon.

Regards

closed 2022-12-08 by JoachimF

Avalonnw, 2023-02-25:
@dresco was tgere a reason to use 3.3v? TFT connector has all 4 needed pins in theory, but vcc is 5v. This module should work with 5v...

dresco, 2023-02-25:
> @dresco was there a reason to use 3.3v? TFT connector has all 4 needed pins in theory, but vcc is 5v. This module should work with 5v...

@Avalonnw I was concerned 5v data output from the module might not be good news for the MCU. However, have just checked the SKR3 schematic + H743 datasheet, and PA9/PA10 do appear to be 5v tolerant I/Os? Please don't just take my word for it though! :grimacing: Would be interested to hear how you get on..

Avalonnw, 2023-02-25:
> 5v tolerant I/Os? Please don't just take my word for it though! 😬 Would

Looks like they are 5v tolerant.
Also, if it's not too much trouble, can you check which pin is wired where?
I always mix them up. TxD on the module goes to PA9 or PA10?