dresco/STM32H7xx #7: SKR3 direction pin Z and Y2 AXIS

JoachimF, 2022-12-06:
Hello,

I have mounted the SKR3 on the router, Y axis ganged and auto square. But I have the Z and Y2 direction pins that are not working, the board is new and never used. The wiring is NC switch between mcu pin and GND. I have tested the drivers on other axis and I have no problem. The pin stay at 0V.

Any idea?

dresco, 2022-12-07:
I'll get a logic analyser hooked up to my SKR3 board and investigate..

dresco, 2022-12-07:
Sorry, another typo, just pushed a fix;

```
diff --git a/Src/driver.c b/Src/driver.c
index 3e41828..49777d3 100644
--- a/Src/driver.c
+++ b/Src/driver.c
@@ -615,7 +615,7 @@ inline static __attribute__((always_inline)) void stepperSetDirOutputs (axes_sig
     dir_outbits.mask ^= settings.steppers.dir_invert.mask;
     DIGITAL_OUT(X_DIRECTION_PORT, X_DIRECTION_BIT, dir_outbits.x);
     DIGITAL_OUT(Y_DIRECTION_PORT, Y_DIRECTION_BIT, dir_outbits.y);
-    DIGITAL_OUT(Y_DIRECTION_PORT, Z_DIRECTION_BIT, dir_outbits.z);
+    DIGITAL_OUT(Z_DIRECTION_PORT, Z_DIRECTION_BIT, dir_outbits.z);
  #ifdef GANGING_ENABLED
     dir_outbits.mask ^= settings.steppers.ganged_dir_invert.mask;
   #ifdef X2_DIRECTION_PIN
```
The outputs look correct to me with that applied?

```
G0 X10
G0 Y10
G0 Z10
G0 X0
G0 Y0
G0 Z0
```

[image]

dresco, 2022-12-07:
Ok, reconfigured for N_AXIS=4 plus ganged Y, still looks good here?

```
G0 X10
G0 Y10
G0 Z10
G0 A10
G0 X0
G0 Y0
G0 Z0
G0 A0
```
[image]

JoachimF, 2022-12-08:
Hello Dresco,

I have updated the SKR3, now the Z axis is moving in both way, but the Y2 is stuck in one direction. I tried to remove the A axis, so Y2 is now on M3, but same result.
Before A axis direction was working.

This is very strange, I have Ganged axis & Auto square, I'll try without auto square.

dresco, 2022-12-08:
Just for reference, these are the symbols I have defined (building latest master with no other updates);

[image]

dresco, 2022-12-08:
Strange, here is my binary - is that any different?
[grblHAL Driver STM32H7xx.zip](https://github.com/dresco/STM32H7xx/files/10186619/grblHAL.Driver.STM32H7xx.zip)

JoachimF, 2022-12-08:
Hello Jon,

My bad! The copper in the direction wire was broken! Thanks for your help, the router is working, now I have to test the dual spindle and the probe.

dresco, 2022-12-08:
> The copper in the direction wire was broken! Thanks for your help, the router is working

Good news! Thanks for the update..

JoachimF, 2022-12-08:
Here is the router !!

[image]

JoachimF, 2022-12-08:
Thanks to you, I use now my 8 port logic analyser :-)

dresco, 2022-12-09:
Cool, glad it's working for you! I keep wondering about putting a diode laser module on mine as a 2nd spindle, would be interested to hear how you get on..

JoachimF, 2022-12-09:
I'am also interested to add a diode as second spindle, but I'm afraid by
dangerous reflexions of this diodes. I have alaready a K40, but limited to
32*22cm.
I Have also hot wire as second spindle, with a second carriage A and B. the
wire move between XZ and AB. I use it to cut wings in polystyrene. I want
to add plasma cutter more than laser diode, because laser is not very
powerfull.

The picture is a router that I made with students, mine is smaller, it's a
root3cnc.
Next steps are rotary cutter on A axis, and vertical hotwire.
The goal is to cut carbon and glass fiber fabric, and machine surf.

closed 2023-11-14 by dresco