dresco/STM32H7xx #8: SKR 3 is not visible to pc

Avalonnw, 2023-01-19:
Hi,
I've purchased an SKR 3 EZ board specifically to run grblhal, however I have this weird issue: the board is not recognized by PC (no device appears). I tried web builder, I tried cube, I tried platformio. It flashes (file gets renamed) and that's it.
If I flash RepRap firmware, a new Com port gets added, so it's not the board or cable.
What am I doing wrong?

dresco, 2023-01-19:
> It flashes (file gets renamed) and that's it.

Hi, the current firmware is not built for use with the preinstalled SDCard bootloader.

I would suggest flashing with DFU. (Hold the boot button, then press and release the reset button, and then release the boot button. You should then see the board enumerate over USB as a DFU device). Once in DFU mode, the firmware can be flashed over USB using your tool of choice (dfu-util, PlatformIO, STM32CubeProgrammer etc).

Please note that flashing over DFU will overwrite the SDCard bootloader. If that is a concern, then it would need some changes to the build process in order to support their original bootloader..

Avalonnw, 2023-01-20:
That was... unexpected. Thanks. Looks like it works.

closed 2023-01-20 by Avalonnw

dresco, 2023-01-20:
> Looks like it works.

Thanks for confirming. Am sure it would be useful to have another build image for the bootloader offset, will add it to the to-do list. Cheers..