dresco/STM32H7xx #9: MODBUS on SKR3

Avalonnw, 2023-01-20:
Am I missing something or there is no MODBUS spindle support for H7 boards?

Avalonnw, 2023-01-20:
> Hi, yes it should be fine - is something I test with here..
>
> What issue are you seeing? There is a conflict with TMC UART usage which I
> still need to commit a fix for.

I can't figure out which pins to use. I am currently on a RTFM stage before
pulling my CNC for a controller upgrade apart. Unfortunately the FM part in
RTFM is not very user friendly :)

dresco, 2023-01-20:
> I can't figure out which pins to use.

Try [this](https://github.com/dresco/STM32H7xx/issues/6#issuecomment-1328084821), is how I hooked my Huanyang VFD up to the SKR3..

Avalonnw, 2023-01-21:
Thanks, I'll try.
I hope my spindle actually supports modbus :) It's mentioned in the manual in passing and I afraid RS485 components might be not populated on the VFD board. Will check tonight.

closed 2023-04-18 by Avalonnw