dresco/STM32H7xx #11: Multiple compiling issues I can't solve

Avalonnw, 2023-02-26:
I am completely stuck. I know I'm doing something wrong, I just can't figure it out...
Just a bit of background:
I have a 3040 CNC with rotational axis add-on and I want it to run as a 4-axis cnc (XYZA). It was running grbl-mega with XYZ before.
I've got an SKR3EZ board with TMC5160 drivers, Huanyang FC1 VFD and an rs485 converter.
I use ioSender as UI.
I am trying to configure and compile the code here in 3 different ways:
1. Web builder compiles everything ok, but trinamic drivers are not recognized (nor SD card). No motion from the machine.
2. I managed to use STM Cube but despite setting TMC5160 drivers they are not recognized. No motion. No 4th axis. No SD card.
3. Platformio is easier, I managed to get TMC recognized, machine moves, however there is no SD card still AND I can't turn modbus vfd on. #define VFD_ENABLE and #define MODBUS_ENABLE seem to make no difference and if I compare resulting bin file with memory of SKR board in CUBE Programmer there is no difference reported.

Is there a complete, definite guide of sorts, on how to make it work???

Also, it seems like there were a lot of changes to drivers and plugins in the main fork that are broken/not compatible with this fork.
I really need help. Please?

dresco, 2023-02-26:
Apologies, have been a bit distracted with a related project & haven't been keeping up to date with upstream commits, will try to get caught up next week.

Can you attach output from `$I` and `$pins` for starters. Cheers..

Avalonnw, 2023-02-26:
Managed to somehow get it compiled from under Platformio when I changed this section in platformio.ini:

`[env:btt_skr_30_tmc5160]
board = btt_skr_30
board_build.ldscript = STM32H743ZITX_FLASH.ld
build_flags = ${common.build_flags}
              ${usb.build_flags}
              ${sdcard.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_30
  -D HSE_VALUE=25000000
  -D TRINAMIC_ENABLE=5160
  -D TRINAMIC_SOFT_SPI
  -D VFD_ENABLE=-1
  -D MODBUS_ENABLE
  -D FANS_ENABLE
lib_deps = ${common.lib_deps}
           ${usb.lib_deps}
           ${sdcard.lib_deps}
           motors
           trinamic
           spindle
           fans
lib_extra_dirs = ${common.lib_extra_dirs}
                 ${usb.lib_extra_dirs}
                 ${sdcard.lib_extra_dirs}
upload_protocol = dfu`

The problem now is when I try to send a spindle command I get Alarm14 state and restart:

`ALARM:14
GrblHAL 1.1f ['$' or '$HELP' for help]
[VER:1.1f.20221101:]
[OPT:VNMZHTS,1024,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,HOME,TC,SED,TMC=15,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32H743]
[DRIVER VERSION:221014]
[BOARD:BTT SKR-3]
[PLUGIN:Trinamic v0.09]
[PLUGIN:MODBUS v0.14]
[PLUGIN:HUANYANG VFD v0.08]
[PLUGIN:SDCARD v1.07]
[G54:0.000,0.000,0.000,0.000]
[G55:0.000,0.000,0.000,0.000]
[G56:0.000,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000,0.000]
[G28:0.000,0.000,0.000,0.000]
[G30:0.000,0.000,0.000,0.000]
[G92:0.000,0.000,0.000,0.000]
[HOME:0.000,0.000,0.000,0.000:0]
[TLO:0.000,0.000,0.000,0.000]
[PRB:0.000,0.000,0.000,0.000:0]
`

$I response:
`$I
[VER:1.1f.20221101:]
[OPT:VNMZHTS,1024,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,HOME,TC,SED,TMC=15,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32H743]
[DRIVER VERSION:221014]
[BOARD:BTT SKR-3]
[PLUGIN:Trinamic v0.09]
[PLUGIN:MODBUS v0.14]
[PLUGIN:HUANYANG VFD v0.08]
[PLUGIN:SDCARD v1.07]`

$PINS:
`$PINS
[PIN:PA4,Reset]
[PIN:PA5,Feed hold]
[PIN:PA6,Cycle start]
[PIN:PC13,Probe]
[PIN:PC1,X limit min]
[PIN:PC3,Y limit min]
[PIN:PC0,Z limit min]
[PIN:PC2,A limit min]
[PIN:PD4,X step]
[PIN:PA15,Y step]
[PIN:PE2,Z step]
[PIN:PD15,A step]
[PIN:PD3,X dir]
[PIN:PA8,Y dir]
[PIN:PE3,Z dir]
[PIN:PD14,A dir]
[PIN:PD6,X enable]
[PIN:PD1,Y enable]
[PIN:PE0,Z enable]
[PIN:PC7,A enable]
[PIN:PD5,Motor CSX]
[PIN:PD0,Motor CSY]
[PIN:PE1,Motor CSZ]
[PIN:PC6,Motor CSM3]
[PIN:PB3,Flood]
[PIN:PB4,Mist]
[PIN:PA10,RX,Modbus]
[PIN:PA9,TX,Modbus]`

Avalonnw, 2023-02-26:
Compared to what you have [here](https://github.com/dresco/STM32H7xx/issues/6#issuecomment-1328084821) my $I does not show Spindle position.

dresco, 2023-02-26:
That's encouraging, Alarm 14 means it's not getting a correct response from the VFD.. That could be wiring, baud rate mismatch, or protocol issues.

Have just checked the way I had my module physically connected to the SKR3, and PA9 (centre pin) on the TFT connector goes to TX on the module.

Baud rate on the grblHAL side is selected with $374, be worth checking that matches the VFD settings.

Your VFD looks like a newer model than mine, so could also be protocol related. There are 2 different Huanyang protocols already supported, plus a bunch more for other VFDs.

If you want to try different spindle configurations at runtime (will give you the SPINDLE option in SI), [this](https://github.com/dresco/STM32H7xx/issues/6#issuecomment-1328050074) comment had the instructions near the end.

Avalonnw, 2023-02-26:
Thank you. I'll try that tomorrow.

Avalonnw, 2023-02-27:
Okay... I tried ALL spindle configurations I could think of, with the same result - Alarm14.
On the VFD there is a 485 jumper that supposedly switches comms on/off. I tried it in both positions.
I can't find any dedicated settings in the manual that would specifically allow control via 485, apart from one setting for a button that would allow control to be switched from manual to external. I tried both settings with all spindle types also...
So, still not working.

Is there a way to see any raw data coming back and forth between VFD and SKR?

[This is a link to VFD manual BTW, just in case.](https://drive.google.com/file/d/1ClaLMfNDkWEd8mPnLNmg_PzwBDrnM0mc/view)

dresco, 2023-02-27:
That looks much better than the earlier Huanyang models. It seems to be properly Modbus RTU compatible, so should be possible to configure it with the new modvfd spindle type, and I think all the needed info is in the manual..

If it's okay with you, I'll open a discussion on the main grblHAL repo, should get a bit more visibility there?

dresco, 2023-02-27:
> Is there a way to see any raw data coming back and forth between VFD and SKR?

I forget, did you say you had a logic analyser? If so, then yes, would be really helpful to put it on the PA9/PA10 connections between the board and module..

Avalonnw, 2023-02-27:
Unfortunately I don't have a logic analiser :( It's one of those things I
alway wish I had but when there is no need I forget about :)

If you could raise the issue in the main branch, that would be great!
Unfortunately there is not much info about these vfds at the moment. I
should've bought the older model :)

dresco, 2023-02-27:
Have raised the issue [here](https://github.com/grblHAL/Plugins_spindle/issues/17), you may want to subscribe for updates..
Cheers. Jon

Avalonnw, 2023-03-13:
Hi Jon,

Still no luck. I have Alarm14 state (no comms?).
P0.01 set to 2.
Port speeds set to 19200 N,8,1
I have set 2 spindles according to [this](https://github.com/dresco/STM32H7xx/issues/6#issuecomment-1328050074), as I don't see any spindle options otherwise.
Left Spindle 0 as PWM, set Spindle 1 as ModVFD (set address to 1)
After M104Q1:
```
M104Q1
$I
[VER:1.1f.20230228:]
[OPT:VNMZHSL,35,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,HOME,TC,SED,TMC=15,SD]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:STM32H743]
[DRIVER VERSION:230129]
[BOARD:BTT SKR-3]
[PLUGIN:Trinamic v0.11]
[PLUGIN:MODBUS v0.14]
[PLUGIN:MODVFD v0.03]
[SPINDLE:MODVFD]
[PLUGIN:SDCARD v1.07]
```
I can see TXD light blinking on RS485 converter (before flashing with updated firmware RXD was blinking also). And that's it.

I wonder if I am compiling it incorrectly or if I'm missing something.

dresco, 2023-03-13:
If you were previously seeing at least the RXD blinking, perhaps worth double checking the comms settings? From looking at the manual, these are the relevant ones that I see;

```
VFD settings
P00.01	2	Enable Modbus control
P14.00	1	Modbus address
P14.01	4	19,200 baud
P14.02	0	N,8,1

grblHAL settings
$460=1	Spindle Modbus address
$374=3	19,200 baud
```

After checking those, might be time for a <$10 logic analyser just to see what is (or isn't) being sent & received?

Avalonnw, 2023-04-14:
Got the analyzer and a different modbus-RS232 board. No luck. Analyzer shows packets are incomplete. [Pulseview file here](https://drive.google.com/file/d/1buddbAazn3SDkwnQIzcmqqncCRFxK54C/view?usp=sharing)

Avalonnw, 2023-04-14:
Okay... I've been checking the VFD with direct connection to PC via Modbus Poll and by manually typing commands I can see the response from VFD is 86 06, which, according to the manual means CRC check fail. Sometimes VFD turns on (via manual command or via ioSender, but iosender gets Alarm14 and everything turns off) - the lights on VFD come on, etc.
I understand the cause for CRC error could be in timing? RTU frames supposed to have some quiet time of 3.5 bytes beforehand?

dresco, 2023-04-14:
Could you try another capture with a faster sample rate? Looks like the sample rate on that one was only 20kHz, so the analysis may be struggling, even at 19,200 baud.

From what I've experienced with the older models, they will just ignore the subsequent frame if the silent time is set too low.

Also, if you could attach your `$I` and `$$` output, I'll duplicate here and at least verify the initial requests look sane. Seems promising if it starts sometimes though.. It's possible that the logic side is okay, but having noise issues on the RS485 side? What sort of cable are you connecting with, a twisted pair?

Avalonnw, 2023-04-15:
Okay... I think I might have narrowed down what the problem is...
I've re-made all the wiring and the behavior is the same as before, although more stable. I've put wires (twisted pair) through toroidal suppressors on both rs232 and 485 ends. Seemed to help. However...
I've been testing the VFD with MODBUS Poll app and I can control Forward/Reverce/Stop by writing 0001/0002/0005 to address 2000 BUT I can see no way of controlling the speed. I had a look at your post [here](https://github.com/grblHAL/Plugins_spindle/issues/17#issue-1601018080) and you mention 2001 register for setting frequency however I can't make it work. In the manual they call it "Communication setting frequency" and it should go in increments of 0.01Hz so when I send a value of 20000 (4E20) as a test I get nothing (just response ok). I was wondering if I missed some setting or something...

dresco, 2023-04-15:
Just checking the manual again, looks like perhaps P00.06 needs to be set to 8

Avalonnw, 2023-04-15:
LEGEND!!! If you pass Adelaide one day, I'll buy you a beer... a case of it!
P00.06 set to 8, P00.07 =2, P00.09=4

closed 2023-04-15 by Avalonnw

dresco, 2023-04-15:
Good news, glad it's working! 

Perhaps also worth documenting the required VFD and grblHAL settings in that post in the Spindle plugin repo? Am sure that will help the next person that picks up one of these new models..

Cheers, Jon.

Avalonnw, 2023-04-15:
Will do, thanks.