dresco/STM32H7xx #12: Second Y motor not works

Demitrius, 2023-03-04:
Dear Jon

I have wire motors to SKR 3 EZ board.
- motor to X port
- another motor to Y port
- another one motor to E0 port

Uncomment lines in my_machine.h
```
#define Y_GANGED            1
#define Y_AUTO_SQUARE       1
```

Build, flash, connect to board by telnet and type

X100 - motor X works
Y100 - motor Y works, but E0 motor not works at all.

Where is my mistake?

dresco, 2023-03-04:
Hi, how up to date is your code - there were some related fixes in Nov/Dec?

Does your `$pins` output show the Y2 step and direction pins as expected?

Demitrius, 2023-03-04:
Code latest, updated today

$pins output

```
[PIN:PA4,Reset]
[PIN:PA5,Feed hold]
[PIN:PA6,Cycle start]
[PIN:PC13,Probe]
[PIN:PC1,X limit min]
[PIN:PC3,Y limit min]
[PIN:PC0,Z limit min]
[PIN:PD4,X step]
[PIN:PA15,Y step]
[PIN:PE2,Z step]
[PIN:PD3,X dir]
[PIN:PA8,Y dir]
[PIN:PE3,Z dir]
[PIN:PD6,X enable]
[PIN:PD1,Y enable]
[PIN:PE0,Z enable]
[PIN:PD5,Motor CSX]
[PIN:PD0,Motor CSY]
[PIN:PE1,Motor CSZ]
[PIN:PB6,Spindle on]
[PIN:PB5,Spindle direction]
[PIN:PB3,Flood]
[PIN:PB4,Mist]
[PIN:PB0,Spindle PWM]

```

dresco, 2023-03-04:
I'd expect to see the Y2 pins listed there too if the #defines had been picked up..

How are you building, CubeIDE?

Note that if using PlatformIO, it ignores my_machine.h, so you would want to add `-D Y_AUTO_SQUARE=1` to the `[env:btt_skr_30_tmc5160]` section (I believe auto square implies ganged, so that one should be sufficient).

Demitrius, 2023-03-04:
Updated platformio.ini
```

[env:btt_skr_30_tmc5160]
board = btt_skr_30
board_build.ldscript = STM32H743ZITX_FLASH.ld
build_flags = ${common.build_flags}
              ${usb.build_flags}
              ${sdcard.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_BTT_SKR_30
  -D HSE_VALUE=25000000
  -D TRINAMIC_ENABLE=5160
  -D TRINAMIC_SOFT_SPI
  -D Y_AUTO_SQUARE=1
  -D Y_GANGED=1

```
$pins contains Y2
```

[PIN:PA4,Reset]
[PIN:PA5,Feed hold]
[PIN:PA6,Cycle start]
[PIN:PC13,Probe]
[PIN:PC1,X limit min]
[PIN:PC3,Y limit min]
[PIN:PC0,Z limit min]
[PIN:PC2,Y limit min 2]
[PIN:PD4,X step]
[PIN:PA15,Y step]
[PIN:PE2,Z step]
[PIN:PD15,Y2 step]
[PIN:PD3,X dir]
[PIN:PA8,Y dir]
[PIN:PE3,Z dir]
[PIN:PD14,Y2 dir]
[PIN:PD6,X enable]
[PIN:PD1,Y enable]
[PIN:PE0,Z enable]
[PIN:PD5,Motor CSX]
[PIN:PD0,Motor CSY]
[PIN:PE1,Motor CSZ]
[PIN:PC6,Motor CSM3]
[PIN:PB6,Spindle on]
[PIN:PB5,Spindle direction]
[PIN:PB3,Flood]
[PIN:PB4,Mist]
[PIN:PB0,Spindle PWM]
```

But only one Y motor works. Y2 wired correctly

Demitrius, 2023-03-04:
I use STM32CubeProgrammer

dresco, 2023-03-04:
Can you attach your binary here, and I'll try it on my board tomorrow? (I definitely had all five motor outputs working last time I tested my SKR3 here, but will double check it all).

Cheers..

Demitrius, 2023-03-04:
[firmware.zip](https://github.com/dresco/STM32H7xx/files/10889312/firmware.zip)
Yes, please check it, may be configuration enough.

Early I test hardware port E0 - just put same params values (dir,step,ena) to X and motor works when X100 command thru port E0

Demitrius, 2023-03-04:
[src.zip](https://github.com/dresco/STM32H7xx/files/10889326/src.zip)
My src configs

dresco, 2023-03-05:
Thanks, have checked the outputs with a logic analyser.

Step and direction signals look correct on all outputs, but the enable signal isn't changing on E0 when it's a ganged axis (looks fine when configured as an independent axis). 

Not sure why yet, but will look into it this week. Thanks for reporting!

dresco, 2023-03-05:
@Demitrius Please try now, have just pushed a [fix](https://github.com/dresco/STM32H7xx/commit/f98e51c92d84bcfaad530213a55c1cd51cc473b6) to master..

@terjeio FYI, believe this will apply to the F7 driver also.

Demitrius, 2023-03-05:
Jon, this fix solve issue!

Both Y motors works like sharm.

Cheers :)

closed 2023-03-05 by Demitrius