dresco/STM32H7xx #13: SKR 3 EZ and wire limits question.

Demitrius, 2023-03-06:
Hello Jon,

I found limits wire scheme, with resistors and capacitor.

Scheme attached.
[image]

What do you think, need wire same for SKR 3 EZ?

In my build, I can connect 2 limits (start and end stops) to X-axis limit port, cable length about 3 meters.
May be better to add capacitor/resister?
[image]

dresco, 2023-03-06:
Looking at the [schematic](https://github.com/bigtreetech/SKR-3/blob/master/Hardware%20(SKR%203%20EZ)/BIGTREETECH%20SKR%203%20EZ%20V1.0-SCH.pdf), they already have the onboard resistors and capacitors for the limit switches.

Ideally all the inputs would come through optocouplers for added protection, but I suppose it's a 3D printer board after all.. :shrug:

Demitrius, 2023-03-06:
Completely agree, the optocouplers is definitely overhead.
I will do it as it is, without extra elements.

Jon, thanks for help. I appreciate it.

closed 2023-03-06 by Demitrius