dresco/STM32H7xx #17: TMC5160 not working

joaobarros, 2023-10-11:
When trying an SKR3 with TMC5160 drivers I get this on startup:
<0D><0A>[MSG:Warning: Could not communicate with stepper driver!]<0D><0A>

BTW, the btt_skr_v3.0_map.h file has this typo, could be related:

// The BTT SKR-3 uses software SPI
#define TRINAIC_SOFT_SPI

dresco, 2023-10-14:
Thanks for pointing that out, is definitely a typo. How are you building?

If using PlatformIO (or the web builder), then it shouldn't actually matter, as `TRINAMIC_SOFT_SPI` is already defined from the platformio.ini file. However, it **will** affect STMCubeIDE builds if you have (for instance) taken one of the existing BTT SKR3 entries and changed `TRINAMIC_ENABLE` to `5160`..

Am currently travelling for a few weeks, so afraid I'll not be able to test anything in the short term..

joaobarros, 2023-11-01:
Hi, I used the webbuilder

dresco, 2023-11-06:
> Hi, I used the webbuilder

Thanks, I'll be back from travelling in a few days, will be able to take a look then..

dresco, 2023-11-13:
Hi, just back home and starting to look at this..

It does seem to be working for me with a local build and a single 5160 breakout board attached via jumper wires (I don't have any 5160 plug-in modules).

Could you perhaps share your web builder config with me (from the "save board" button)? Thanks!

dresco, 2023-11-14:
btw, have pushed a fix for the typo in the first comment, thanks for bringing that to my attention!
(wouldn't impact the webbuilder images though, so not the cause of your issue)

dresco, 2023-11-15:
> (wouldn't impact the webbuilder images though, so not the cause of your issue)

Actually, I take that back. On further thought I believe it **would** have affected webbuilder images (just not local platformio builds). Will get that updated..

closed 2024-05-25 by dresco