dresco/STM32H7xx #18: H723 ZET6 compatibility?

JoeCorelli, 2024-01-08:
Hi, 

Looking to incorporate a Bigtreetech Octopus MAX EZ with an H723 ZET6 chip into my CNC project. Wondering if simply remapping pins would allow use of the board with grblHAL and if not how could I request implementation. Apologies if this is not the correct place for the question.

dresco, 2024-01-09:
Hi, yes it's the right place to ask..

Just been having a look at the schematic, a few things of note:

Processor and clock setup should be fine, is as per the other H723 boards.

What Trinamic drivers are you using? 2209 and 5160 should be fine.

It would be the first 512KB flash device supported for this driver (the rest have been 1MB or 2MB), so would need a new linker file. 

Don't think flash size should be a problem (from memory, a basic build is I think around the 250KB mark). We lose 128KB of flash for the SDcard  bootloader, and would need to make sure the I2C EEPROM support is working (not yet tested) - as couldn't spare another 128KB for flash EEPROM emulation.

The SDcard slot wouldn't be usable without code changes, the H7 boards so far have all used SDIO, whereas this one uses a SPI connection.

Networking support has been limited to the onboard Ethernet peripheral, so if you're wanting to use the WiFi module then that would be an unknown..

In summary, with the necessary linker changes, and a board map file (and as long as EEPROM works as it should for settings), then I think it should have basic functionality.

If you want to attach your board map file, I'll have a go at the linker side..

JoeCorelli, 2024-01-09:
[Octopus_MAX_EZ_V1.0_map.txt)

Not sure if my map is 100% correct, but I think it should be ok with the exception of maybe ports for I2C, Serial, etc..  Yesterday was the first time I've ever looked at any of this stuff. 

I will be using BTT TMC 5160 EZ Drivers

My plan is to run the board via USB via Raspberry Pi, so I do not believe the wifi module or ethernet will be necessary on the board.

Also, I plan to flash via DFU which should eliminate need for the SD Card and Bootloader?

Let me know if you need anything from me.  Willing to work for it!

JoeCorelli, 2024-01-09:
Doesnt look like its uploading. Here is a link.
https://github.com/JoeCorelli/grblHAL_config/blob/main/Octopus_MAX_EZ_V1.0_map.h

dresco, 2024-01-09:
Thanks! I started on the build support today, so will aim to push a test branch tomorrow..
Cheers,
Jon.

JoeCorelli, 2024-01-09:
Awesome. Thanks a lot!

JoeCorelli, 2024-01-10:
Hey Jon,
Just wanted to add that I pulled a few files from the Marlin source code for this board that may (or may not) help you with getting this board configured.
[https://github.com/JoeCorelli/grblHAL_config/tree/main/Sample_Files](url)

dresco, 2024-01-10:
Hi Joe,

Have pushed a test branch [here](https://github.com/dresco/STM32H7xx/tree/octopus)

Am assuming you have a build environment set up already (PlatformIO is easiest), else I can attach a binary here?

In the PlatformIO config, I've added two new build environments for the Octopus board. One for 5160 drivers without bootloader, and one for 2209 drivers with bootloader (these are mainly for testing coverage, as they all get built on push to master).

Have kept all the ports the same, only thing to note is that SPI4 is used for the Trinamic drivers, which I don't believe can be shared with other SPI devices in grblHAL. I notice SPI3 is exposed on the WiFi headers, so that would be an option if needed for something else..

I think it would be worth keeping en eye on the 5160 current draw on initial testing, as had another SKR3 user fry some drivers.. They should be set to a sane level at startup (check with `M122`), but if you see a communication error message, then I'm not sure what they may end up trying to draw.

Cheers,
Jon

JoeCorelli, 2024-01-10:
Jon,

Thanks for the quick turnaround. I have not used PlatformIO, but I will give it a try.

Will keep an eye on the driver current. I have 2209 drivers as well, so I can test those as well.

Thanks,
Joe

dresco, 2024-01-10:
> Thanks for the quick turnaround. I have not used PlatformIO, but I will give it a try.

The PlatformIO builds override any settings changes in my_machine.h, so configuration is done through the  `platformio.ini` file. See one of the [newly added](https://github.com/dresco/STM32H7xx/blob/2e72aab0ef6d6ae6cb55b32c6f253ebf05fc2a38/platformio.ini#L346-L364) environments for a starting point.

I will add an Octopus build to the STM32CubeIDE configuration as well.

Once we know it's working, I'll get the H7 [Web Builder](http://svn.io-engineering.com:8080/) configuration files updated for Terje..

JoeCorelli, 2024-01-10:
ok great!
It looks like I was able to figure it out.  Going to flash the firmware in a few minutes.

JoeCorelli, 2024-01-10:
Jon,
Jon,

Jon,  
5160 Firmware flashed and board connected using UGS. 

UGS will not allow any console commands until I have all switches hooked up, so I will work on getting those connected to continue testing. I may just place jumpers on the required contacts to access configs temporarily.

Will get back with you once I make some progress.

Thanks again for the quick turnaround.

Joe

JoeCorelli, 2024-01-11:
Jon,

I was able to get the board connected to UGS and IOSender, but its not communicating with the drivers. Not entirely sure if its the firmware I built or if there may be another issue with my settings.

 I attached a link to pictures of my settings/errors from IOSender.

https://github.com/JoeCorelli/grblHAL_config/tree/a70bf99372c00bcfaf2b549497b6c6b3f2572407/ERRORS

dresco, 2024-01-11:
Actually, on checking the driver code, only SPI1-SPI3 were being initialised. 

Have just pushed an update to include SPI4 if you'd like to try again?

Cheers,
Jon.

JoeCorelli, 2024-01-12:
Jon,

After adding your suggestions into the new build I'm still getting the same error with m122, but it looks like I can adjust current in the settings, which is set at 500mA by default and should not cause any issues with any of the 5160 driver variants. Mine are rated up to 4.5A (peak) and have a large heat sink (BTT EZ 5160RGB).

I am able to get movement on the z axis motor when homing the machine,, but I do not have physical endstops installed at the moment., so it does not progress past z homing cycle to check the other motors/drivers. 

I have not attempted configuring the board for using sensorless homing YET although it will make the conversion easier for my machine which currently uses stall homing and has no endstops. Are any additional firmware changes required for this or is it a simple config change in IOSender? 

I am going to get the controller configured into more of a functional state before I continue testing and should make progress next week. 

Good news is all switch and alarm functions seem to be working as defined in the map.h file and all of the smoke is still conatained in the control board. I will keep you posted with progress.

Joe

dresco, 2024-01-12:
Mostly good news then ;)

While troubleshooting - personally I would enable Trinamic support for a single axis at a time with the $338 setting, and work up from there. Do you have access to a logic analyser or scope to check the signalling? I've not checked it yet here, will do that today..

Afraid I've not really got any suggestions for sensor-less homing, probably best off asking in the [Plugins_motor](https://github.com/grblHAL/Plugins_motor) repo.

Cheers,
Jon

JoeCorelli, 2024-01-12:
I do not have a scope or analyser. This is my first dive into any project this deep, so I'm learning as I go and should probably have more tools and knowledge for this kind of work. 

For now my goal is to get basic functionality and then fine tune and add more features. Still need to work out the spindle (huanyang 2.2kw) and laser functions as well. 

I dug out an old 3018 I have and will be testing with that before I move to my bigger machine.

JoeCorelli, 2024-01-13:
Jon,

I played around with the build today and I think I may have built the firmware incorrectly.  
After adding -D TRINAMIC_SPI_ENABLE=1 to platformio.ini I get the Trinamic Tuner tab in IO sender and after clicking "get status" I get this:  [image]

It seems as though the drivers are connected.  M122 does nothing but no error is present either?

Also unsure how to enable ABC axis motors, but it seems like N_AXIS_MOTORS=X should be defined somewhere. I can't find it besides in "#ifdef"  configurations (the build fails if put -D N_AXIS_MOTORS=1 into platformio.ini).  I'm not familiar with the file structure of the repo or how dependencies work with grblHAL, so its slow going...any pointers are welcome if you have the time. 

Joe

dresco, 2024-01-13:
Hi Joe,

`TRINAMIC_SPI_ENABLE` should get set automatically when `TRINAMIC_ENABLE=5160`, so sounds like something isn't quite right. Can you share your platformio.ini file?

That "get status" output you've shared comes from the `M122` command, so that all looks fine. But are you saying you don't see that output when you type it in? (I don't really use IOSender, other than occasional testing in a VM, as don't have any Windows machines)..

Total number of axis can be set with` N_AXIS=n`. If needed, each ganged/auto-squared axis can be enabled with (for instance) `Y_GANGED=1` or `Y_AUTO_SQUARE=1`.

Although `my_machine.h` isn't used by the PlatformIO builds, it's a good place to look for the available options. Other docs & wiki are in the main grblHAL [repo](https://github.com/grblHAL/core).

I'll get the [Web Builder](http://svn.io-engineering.com:8080/) files updated next week, that should make it a lot easier to explore various configurations for testing..

Cheers,
Jon.

JoeCorelli, 2024-01-14:
Hey Jon,

The web builder will make my life a bit easier. Cant wait!

`TRINAMIC_SPI_ENABLE` should get set automatically when `TRINAMIC_ENABLE=5160`, so sounds like something isn't quite right.' 

I looked around trying to follow the trinamic reference path and traced it back to grbl/driver_opt.h and it is defined there but it seems like it only works when it wants to. Stepper motor settings are still available and motors drive (most of the time).

'That "get status" output you've shared comes from the `M122` command, so that all looks fine. But are you saying you don't see that output when you type it in? (I don't really use IOSender, other than occasional testing in a VM, as don't have any Windows machines)..'

I was able to use M122 once which gave the same info, but after re-flashing only adding  -D Y_GANGED =1  both the tab and communication disappeared again, and removing it again had no effect. Was never able to get it to come back.

 Which sender do you use? I am only using IOSender at the moment because I thought a RPi5 would be a good idea and most of the senders I have tried to use are not compatible yet. I did get UGS to load but I get nothing as far as driver configuration, so I went with IO. 

'Total number of axis can be set with` N_AXIS=n`. If needed, each ganged/auto-squared axis can be enabled with (for instance) `Y_GANGED=1` or `Y_AUTO_SQUARE=1`.'

Ganged axis worked fine tested with Y and both Y and A ran together and N_AXIS did the trick However, still no M122 with the communication error.  N_AXIS=5 decouples Y and A although Y_GANGED is still set.  This reverts to XYZAB instead with all motors moving independently.
 
Going to pick this back up once the web builder is available, unless you need me to check anything before hand.
Joe

dresco, 2024-01-14:
Hi Joe,

[Web Builder](http://svn.io-engineering.com:8080/) has been updated now, give that a try..

> Which sender do you use? 

On my router, am using CNCjs (on a headless Pi). Wouldn't necessarily recommend it though, as doesn't seem to have much development these days. I would certainly use IOSender if we were a Windows household, just a pity it's not cross platform..

And for development work, mostly just a serial terminal for configuration and basic movements.

> N_AXIS=5 decouples Y and A although Y_GANGED is still set. This reverts to XYZAB instead with all motors moving independently.

I think that sounds right. As I understand it, the ganged motors will be **after** all the defined axis motors. So in that case you would have X, Y, Z, A(M3), B(M4), and **then** your ganged Y motor on C(M5). The `$pins` command should show you what signals are on what pins.

Cheers,
Jon.

JoeCorelli, 2024-01-14:
Jon,

Thanks for getting this out so quickly!

I tried a couple configurations using the web builder with no luck. 

Starting from scratch using only XYZ with everything else disabled there is no communication with the drivers and I get no motion.

Here is the .json from the web builder.  [(https://github.com/JoeCorelli/grblHAL_config/blob/main/STM32H7xx_BTT_OCTOPUS_MAX.json) ](https://github.com/JoeCorelli/grblHAL_config/blob/main/STM32H7xx_BTT_OCTOPUS_MAX.json) 

It looks like when enabling functions it assigns =1,=2=3 for the selected items vs all being =1 (e.g TRINAMIC_SPI_ENABLE=2 in the .json)  Not sure if that is correct for the web builder config or if it is actually assigning =2,=3, etc. in the firmware.   

Joe

dresco, 2024-01-14:
> It looks like when enabling functions it assigns =1,=2=3 for the selected items vs all being =1 (e.g TRINAMIC_SPI_ENABLE=2 in the .json) Not sure if that is correct for the web builder config or if it is actually assigning =2,=3, etc. in the firmware.

Yeah they look correct to me, the values are to do with the SPI chip select options and the EEPROM size.. 

I'll need to have a think about what is going wrong though. Did you say you also had 2209 drivers, are you able to sanity test with those?

JoeCorelli, 2024-01-14:
yes I have them ill plug those in later and see what happens.

dresco, 2024-01-14:
In the meantime, I've ordered some more EZ5160 Pro modules (I only had a single 5160 module for testing).

JoeCorelli, 2024-01-15:
Hey Jon,

Same story with the 2209 drivers. If left in a 3 axis (no ganged axis... spindle and ModBus enabled ) configuration drivers work fine. However...  

- Once you set Y_GANGED with 3 Axis selected, you lose communication with drivers and Trinamic tuning/M122 function goes away.  Tried moving drivers/motors between A,B and C to see if it was mapping it to a different driver but got nothing. No other features or plugins were enabled.

- If set to 4 Axis (no ganged axis) same result no communication. The A Axis is shown, but does not work. The strange part is when A is jogged the Z motor turns on but does not move. A Axis is completely inoperative.

Seems like the reference to driver/axis mapping for ABC Axis is missing.

I ordered some EZ5160 Pros today as well.  I currently have 4 EZ5160RGB which are fairly new and not exactly sure if they are causing issues. (same 5160T chip but can run up to 48v and ~3.5A(RMS)

JoeCorelli, 2024-01-15:
Jon,
> What's the end goal with the hardware setup - 3 axis with ganged Y, or something more?

My goal is 4-Axis with ganged Y.  I'd like to add a rotary axis at some point.

> When you set either N_AXIS=4, or Y_GANGED, do the pin mappings look like they line up with the documentation? (See $pins output)?

All of the pins are correct. The only thing I see is what I'm assuming should read "Y2 Enable" is a second Y Enable. (Pin PB6 - M4 enable pin) This is in the N_AXIS=4 and Y_GANGED $pins result.
[image]

I was able to get movement on XYZA but nothing on Y2 (M4) Turns out I have a dead 2209 driver, which may be an issue. If you have an axis enabled without a functioning driver installed does Trinamic Tuning go away? If so, it would explain why setting anything over single motor XYZ caused issues, considering the driver I had in the M3 position was bad and I only have 4 drivers not in use at the moment.

I configured for Y_GANGED and Y_AUTOSQUARED then Z_GANGED and both return the second Y or Z enable which kind of makes sense now but I'm unsure if the ganged axis' enable pin is being inverted or not. Ill pull some 2209s out of my 3d printer tomorrow and see if I can get things going with all 6 drivers in there. ( should have been step 1 I know )

dresco, 2024-01-15:
@terjeio  Seems to be a bug in driver.c, inherited from the F7 driver. 

The ganged axis step outputs are only updated if squaring is enabled. (I think that's how I'd missed it before, as I'd normally select auto-squared rather than just ganged).

https://github.com/dresco/STM32H7xx/blob/69afce6c3c773aa0d935d64958804345a8a4c423/Src/driver.c#L539-L541

dresco, 2024-01-15:
Thanks Terje, will keep an eye out and update the H7 driver to match..

JoeCorelli, 2024-01-15:
After some more testing today I found that with 2209 drivers and motors connected to `X,Y,Z,M3,M4,M5` everything is working with the exception of M5. Whether configured as C axis or Y2 it does nothing, although `$pins` shows correct info .

I did try swapping driver from M4 to M5 and M4 is still working.  Additionally, when jogging the C Axis the B Axis motor is enabled and clicks but does not move.

  Adjusting config to `N_AXIS=4` with `Y_GANGED` moved Y2 to M4 position as expected and it functions normally.  Same result for `N_AXIS=3` with `Y_GANGED`. 

Using `Y_GANGED` or `Y_AUTOSQUARE` works in all configs as well. (except when Y2=M5)

 I'm confident the combination of bad driver and not having everything connected caused the majority of issues I have been seeing with the 2209 drivers.

Joe

terjeio, 2024-01-15:
> Using Y_GANGED or Y_AUTOSQUARE works in all configs as well. (except when Y2=M5)

Hmm, there is another bug it seems, in [this preprocessor code](https://github.com/grblHAL/core/blob/685da469cb2bfc9c6e024785e9f18c4042bb8712/motor_pins.h#L588-L596) that is later used to resolve axis to motor binding (only used by the Trinamic code). `N_AXIS` in the expressions should be replaced by `3` I think. Or remove `N_AXIS` and change `- 1` to `+ 2` to make it simpler.
If I am correct then step/dir outputs are working but not the driver communication.

JoeCorelli, 2024-01-15:
I think I found the issue. in [btt_octopus_max_map.h](https://github.com/dresco/STM32H7xx/blob/master/Inc/btt_octopus_max_map.h) M5 enable pin should be PD5 vice PG9 which is the CS Pin. Verified CS pins for UART and SPI are correct. 
[image]

JoeCorelli, 2024-01-15:
Jon, Terje,

I figured out the issues with both 2209 and 5160 drivers. I have sent you chasing ghosts for long enough. (sorry)

Changing M5 enable pin to the correct pin PD5 fixes the issues with M5
[image]

For the 5160 Drivers somehow the software SPI reference dropped out of the [btt_octopus_max_map.h](https://github.com/dresco/STM32H7xx/blob/master/Inc/btt_octopus_max_map.h)  file, so adding it back in fixes the issue.
[image]

I have motion on all 6 AXIS including Y_GANGED function now for both 2209 and 5160 Drivers along with Trinamic Tuning/M122.

Not sure if there is anything else you need me to check but if there is, we can finally proceed past basic function.

 I do still need to figure out the spindle and if RS485 is an option with this board?

dresco, 2024-01-16:
Hi Joe

> Changing M5 enable pin to the correct pin PD5 fixes the issues with M5

Good spot, thanks. Will get that merged..

> For the 5160 Drivers somehow the software SPI reference dropped out of the [btt_octopus_max_map.h](https://github.com/dresco/STM32H7xx/blob/master/Inc/btt_octopus_max_map.h) file, so adding it back in fixes the issue.

Yeah I'd removed that deliberately, as is wired to the hardware SPI4 peripheral on this board. Interesting that it works bit banged but not using hardware SPI though. Will have a look at any timing differences here..

I assume you are compiling locally to test? There is one difference that does immediately spring to mind - could you try changing the slew rate in the hardware SPI4 setup. Change the speed from `GPIO_SPEED_FREQ_VERY_HIGH` to `GPIO_SPEED_FREQ_LOW` in the code below;

https://github.com/dresco/STM32H7xx/blob/9da16616ab98312747eaae0e72cacbbefee2845a/Src/spi.c#L162-L172

> I do still need to figure out the spindle and if RS485 is an option with this board?

RS485 should work fine, you'll need a transceiver board wired to the UART pins in the map file. I already added UART3, which is wired to TFT header (pins PD8 & PD9), but there is also UART7 wired to the WiFi header (pins PE7 & PE8) if more convenient.

JoeCorelli, 2024-01-16:
> I assume you are compiling locally to test? 

Yes once I found M5 enable pin was wrong.

> could you try changing the slew rate in the hardware SPI4 setup. Change the speed from GPIO_SPEED_FREQ_VERY_HIGH to GPIO_SPEED_FREQ_LOW

I will check this tomorrow.
Guessing I should remove software spi to see if it allows connection with hardware? Also are there any other freq settings? High/Medium? In case low doesn't work.

Did you see my question about sense resistors? Is that a question for Terje?

For rs485 I did some diging around and reached the same conclusion. Got an rs485 module coming then on to wiring.

terjeio, 2024-01-16:
Sense resistors values are defined[ here](https://github.com/grblHAL/Plugins_motor/blob/294c801896f5811e2e442c67ae8f5bf63e4af2ab/trinamic.h#L33-L44). I should make these overridable from the board map?

dresco, 2024-01-16:
> Guessing I should remove software spi to see if it allows connection with hardware? 

Yup, just comment out the `TRINAMIC_SOFT_SPI` definition to revert to hardware SPI.

> Also are there any other freq settings? High/Medium? In case low doesn't work.

Good guess ;) There are low, medium, high, v high... 

Low is used for the soft SPI implementation, and I think should be good up to a couple of MHz. (The Trinamic SPI is configured at 1.5MHz).

dresco, 2024-01-16:
> I should make these overridable from the board map?

I guess the issue is that there are multiple values for 5160 depending on driver package. So would need to be a build time option for different hardware?

JoeCorelli, 2024-01-16:
@terjeio,

> This way it can be overridden from the compile command line, in my_machine.h or even the board map if practical. And I'll have to add drop down entries in the Web Builder for the different variants.

Is it possible to just add the sense resistance value to the driver settings with a default of 0.1ohms and user can change the value as you would for driver current? 
I think it would make it easier in the end. That way you don't have to keep adding variants to the webbuilder or make changes to my_machine.h if someone uses a driver from a different manufacturer.

Joe

JoeCorelli, 2024-01-16:
Jon,

Tested `.Speed = GPIO_SPEED_FREQ_VERY_HIGH` in LOW, MEDIUM and HIGH with software SPI disabled. No change.

When looking around I found this When hovering over SPI_PORT: 

[image]
Not sure if this is just an example, or if the `Expands to: 3` portion is code actually limiting the number of SPI ports and not allowing `SPI_PORT = 4` to function properly? I couldn't find where this is located to try changing it.

dresco, 2024-01-16:
> Tested .Speed = GPIO_SPEED_FREQ_VERY_HIGH in LOW, MEDIUM and HIGH with software SPI disabled. No change.

Okay, thanks for checking..

> Not sure if this is just an example, or if the `Expands to: 3` portion is code actually limiting the number of SPI ports and not allowing `SPI_PORT = 4` to function properly? I couldn't find where this is located to try changing it.

No, it should expand to the actual value for the selected build environment, but sometimes the PlatformIO/VSCode combination seems to get confused (a restart always fixes it for me).

If you right click on `SPI_PORT` there, and "Go to definition", it should be set to 4 early in the map file. You could also verify that the `$pins` output looks correct (PE12/13/14).

I'm only able to test hardware vs software SPI with a dev board (as the SKR3 doesn't have the Trinamic drivers wired to h/w SPI pins), but they have both worked interchangeably for me on my test setup. I'll check some more tomorrow, feel like I must be missing something obvious!!

JoeCorelli, 2024-01-16:
> If you right click on SPI_PORT there, and "Go to definition", it should be set to 4 early in the map file.

Well that made it easier to find!

It took me to reference_map.h? I changed it to 4 in reference map ant it changed the popup as well as made code in SPI 4 Section active? (not grayed out). Recompiled and it still doesn't work. SPI_PORT in Octopus....map.h is set to 4.

`$pins` 
[image]

Seems to be 2 SCK pins identified PB10 and PB11 are for I2C but should PB10 be SCL?

[image]

dresco, 2024-01-16:
> It took me to reference_map.h?

That's taking you to the Nucleo reference map, so doesn't sound like you have the right PlatformIO build environment selected? By default they are all built, but the IntelliSense code help won't match what you're expecting..

Bottom of the screen in the middle;
[image]

Click on that, and will present the list of environments. Then select whichever one has your platformio.ini changes;
[image]

> Seems to be 2 SCK pins identified PB10 and PB11 are for I2C but should PB10 be SCL?

Not an error as such, just interchangeable names for the clock signals. Both the SPI and I2C clocks are labelled SCK in the core code.

JoeCorelli, 2024-01-16:
Yep. Thats what I have selected.

[image]

Still takes me to the reference map though.

dresco, 2024-01-16:
That's weird. Does manually forcing it with Ctrl-Shift-P -> Rebuild IntelliSense index make any difference?

JoeCorelli, 2024-01-16:
I'll check when I get home. Only the Octopus map.h does this. Checked map files for other boards and they aren't referenced to anything.

JoeCorelli, 2024-01-17:
> Does manually forcing it with Ctrl-Shift-P -> Rebuild IntelliSense index make any difference?
No same result.

I decided the best way to see if something was tying spi.c to reference_map.h was to delete the reference map.  After deleting, the SPI reference switched to the Octopus map.h, but after flashing it still did nothing. 
Once I enable software SPI with no other changes everything works 100%. 
Is there any reason why using software SPI is a bad idea? 

Looking at the schematic. The SPI signal goes through U18 and U19 Is that still considered hardwired to the MCU or are they actually software controlled?
[image]

Just throwing out ideas at this point. I'm not sure where to go next.

dresco, 2024-01-17:
Hi Joe,

Think I have finally worked it out..

I'd overlooked that SPI4 has a different clock source to SPI1/2/3, and was running faster than expected (and too fast for the Trinamic modules). 

Can you try this test firmware before I push any changes please? Thanks!!

[firmware.zip](https://github.com/dresco/STM32H7xx/files/13963793/firmware.zip)

Is built for 5160 drivers, hardware SPI, 5 axis + ganged y (and has the M5 enable fix).

JoeCorelli, 2024-01-17:
That was it. Firmware worked great!

dresco, 2024-01-17:
> That was it. Firmware worked great!

Cool, thanks for testing. Will finish testing & push the update tomorrow..

terjeio, 2024-01-17:
> Strange thing is that M122 shows B2 instead of Y2. The B2 motor still moves for the Y Axis though.

Can you try with replacing [this code](https://github.com/grblHAL/core/blob/685da469cb2bfc9c6e024785e9f18c4042bb8712/motor_pins.h#L588-L596) with:

```
#ifdef X_DOUBLED
#define X2_MOTOR (X_DOUBLED + 2)
#endif
#ifdef Y_DOUBLED
#define Y2_MOTOR (Y_DOUBLED + 2)
#endif
#ifdef Z_DOUBLED
#define Z2_MOTOR (Z_DOUBLED + 2)
#endif
```

> Is it possible to just add the sense resistance value to the driver settings with a default of 0.1ohms and user can change the value as you would for driver current?
I think it would make it easier in the end. That way you don't have to keep adding variants to the webbuilder or make changes to my_machine.h if someone uses a driver from a different manufacturer or variant from the same manufacturer. Lately it seems like a new variant comes out every few months.

It is but I'll go for the symbol first as a default value has to come from somewhere and that should be overridable anyway. And I may have to change axis setting number allocations for the [S-curve acceleration](https://github.com/grblHAL/core/discussions/384) profile since this belongs to the core.

JoeCorelli, 2024-01-17:
Terje,
That fixed it. I configured for 4 and 5 Axis and it shows Y2 now.
[image]
[image]

Thanks
Joe

JoeCorelli, 2024-01-21:
Jon,

Any luck integrating those changes?

Joe

dresco, 2024-01-22:
Apologies, bit slow in testing. Master branch has been updated with the latest fixes now..

JoeCorelli, 2024-01-24:
Thanks Jon. 

I'm getting the controller wired up and getting everything ready to test next week. There will probably be some changes to the pin map for the spindle configuration as I don't think I can use the fan pin for enabling the spindle. 

Also would like to use the fans plugin, but not sure what to define in the board map to accomplish that.

tfali2679, 2024-04-29:
Hi, 

I recently picked up a BTT Octopus Max EZ with hopes of getting grblhal to run on it. The issue i'm facing is that no matter what i do, ie use the web builder or compile in STMIDE, I'm unable to get windows to recognize my board after uploading the firmware. I've tried uploading with and without BL using the respective versions of the FW. The only firmware file that seems to allow me to connect to windows and subsequently UGS is the one you provided here:

https://github.com/dresco/STM32H7xx/issues/18#issuecomment-1895831854

I would have no issues using it, however its configured for 5 axis and does not have the fixes made since you created it. 

I would be grateful for any help you could provide.

Some information about my intended build:

3 axis (5 axis planned in future)
on/off spindle
cartesian kinematics
I was also hoping to use external stepper drivers, I don't know if that would be a problem.

Thanks,

Rick

dresco, 2024-04-30:
Hi Riq,

> Edit: Looks like I've made some progress on getting the board recognized.... Could this be because when using "PWM" spindle, I have no spindle attached and that causes some error?

Won't make any difference that a spindle isn't connected, but could be a bug in the code or configuration. I don't have an Octopus Max board here, but do have other H723 boards, so will try and troubleshoot the Octopus build on one of those this week..

> I was also hoping to use external stepper drivers, I don't know if that would be a problem.

No, shouldn't be a problem. Looks like the step/direction/enable signals are brought out on 0.1" headers next to the EZ sockets..

Thanks for the testing/input.
Cheers, Jon.

dresco, 2024-05-25:
For info -  the Spindle PWM fix for the Octopus board is now available in the [Web Builder](http://svn.io-engineering.com:8080/)

Closing this issue as I think that was the last outstanding issue for the currently supported H723 boards.

closed 2024-05-25 by dresco