dresco/STM32H7xx #21: Problem with $X command

PaONEkaruchola, 2024-02-22:
I am using a Nucleo-H743ZI board, I want to run it with grbl firmware, having an issue while killing the alarm, $X is not working, and showing errors.

dresco, 2024-02-22:
Will need some context.. 

What build configuration?
What are the alarms / errors?
Are control and limit inputs inverted as necessary for your connections?

The realtime command `?` will show you the current alarm code and active pin states. There are also the `$LEV` & `$LIM` commands;

```
$LEV - output last control signal events.
$LIM - output current limit pins.
```

PaONEkaruchola, 2024-02-22:
I am using GrblGru V5.1.18 software to send g-code to the controller. While turning off the alarm use the $X command it is not working. 
[image]

dresco, 2024-02-22:
You will need to establish the cause of the alarm. Have you checked the state of the control and limit inputs as described above?

Note that grblHAL expects normally closed switches by default. Most likely cause (on a bare dev board) is that these inputs have not been inverted as needed in the settings..

closed 2024-05-25 by dresco