dresco/STM32H7xx #24: "GRBLHAL - It died on my homemade H723ZGT6 minimal experimental circuit board."

hanke-cnc, 2024-03-07:
"I suspect that some components of my circuit board, not present in the BTT SKR 3, led to its demise, as I had previously run this on my homemade minimal system H750 control board, and it also died. I later found it was due to the RTC program; commenting it out made everything run perfectly. However, this time the H723 experiment didn't go very smoothly. The source code remains unchanged."

[image]
[image]

hanke-cnc, 2024-03-07:
"Since I'm not proficient in debugging, I created a function called led_blink();, which is a function to toggle an LED. I used this to observe where the problem might be occurring. Experimentation revealed that both nvs_buffer_init(); and settings_init(); caused the demise of GRBLHAL."
[image]
[image]

hanke-cnc, 2024-03-07:
"Okay, after commenting out nvs_buffer_init(); and settings_init(); functions, I proceeded further. However, even gc_init(); caused its demise. Okay, after commenting out gc_init();, I finally saw the familiar GRBLHAL messages in the serial terminal. But it died again after sending out the messages."
[image]

hanke-cnc, 2024-03-07:
[image]

hanke-cnc, 2024-03-07:
[image]

dresco, 2024-03-07:
> Experimentation revealed that both nvs_buffer_init(); and settings_init(); caused the demise of GRBLHAL.

Are you using an EEPROM or NVS emulation? If the latter, would imply a mismatch between the hardware (flash layout) with the linker file in use. Is your code anywhere I can see it to check?

hanke-cnc, 2024-03-08:
> > Experimentation revealed that both nvs_buffer_init(); and settings_init(); caused the demise of GRBLHAL.
> 
> Are you using an EEPROM or NVS emulation? If the latter, would imply a mismatch between the hardware (flash layout) with the linker file in use. Is your code anywhere I can see it to check?

Oh my goodness, I've reviewed the photos of BTT SKR 3 again, it uses H723VGT6, while I'm using H723ZGT6, I'll try again, thank you.

hanke-cnc, 2024-03-08:
> > Experimentation revealed that both nvs_buffer_init(); and settings_init(); caused the demise of GRBLHAL.
> 
> Are you using an EEPROM or NVS emulation? If the latter, would imply a mismatch between the hardware (flash layout) with the linker file in use. Is your code anywhere I can see it to check?

Hi Dresco, after I commented out the contents of the memcpy_from_flash and memcpy_to_flash functions, it became normal. I can now communicate with it properly.  🍻

closed 2024-03-08 by hanke-cnc

dresco, 2024-03-08:
> Oh my goodness, I've reviewed the photos of BTT SKR 3 again, it uses H723VGT6, while I'm using H723ZGT6, I'll try again, thank you.

Hmm, well they're both 1MB flash parts, so the failure is a bit of a puzzle. I do plan to include a H723ZG linker map file for completeness, but the contents would be identical to the H723VG file (is only the pin count that's different).

I'm working on a H723VGT6 part here also, but using EEPROM/FRAM for settings on my board. I'll test the flash NVS emulation when I get a chance..