dresco/STM32H7xx #25: Errors and Warnings during compile

gxdeange, 2024-03-08:
Hi dresco,

I cam across some erros and warnings when compiling for 'Release 743 Nucleo'. I have selected BOARD_MY_MACHINE and have created MY_MACHINE_MAP.h file as well. I will be using the HSI (which I have configured in STM32CubeIDE) so I can send any files that can be incorporated into this build.

**Errors:**

Description	Resource	Path	Location	Type
make: *** [Src/subdir.mk:61: Src/spi.o] Error 1	grblHAL Driver STM32F756		 	C/C++ Problem
make: *** [webui/subdir.mk:43: webui/commands_v2.o] Error 1	grblHAL Driver STM32H7xx		 	C/C++ Problem
make: *** [webui/subdir.mk:43: webui/commands_v3.o] Error 1	grblHAL Driver STM32H7xx		 	C/C++ Problem
make: *** [webui/subdir.mk:43: webui/flashfs.o] Error 1	grblHAL Driver STM32H7xx		 	C/C++ Problem
make: *** [webui/subdir.mk:43: webui/fs_embedded.o] Error 1	grblHAL Driver STM32H7xx		 	C/C++ Problem
make: *** [webui/subdir.mk:43: webui/fs_handlers.o] Error 1	grblHAL Driver STM32H7xx		 	C/C++ Problem
make: *** [webui/subdir.mk:43: webui/login.o] Error 1	grblHAL Driver STM32H7xx		 	C/C++ Problem
make: *** Waiting for unfinished jobs....	grblHAL Driver STM32H7xx		 	C/C++ Problem
undefined reference to `_EEPROM_Emul_Start'	flash.c	/GRBL Driver STM32F401/Src	line 41	C/C++ Problem
undefined reference to `_EEPROM_Emul_Start'	flash.c	/GRBL Driver STM32F401/Src	line 46	C/C++ Problem
'sck' undeclared (first use in this function)	spi.c	/grblHAL Driver STM32F756/Src	line 167	C/C++ Problem
'sdi' undeclared (first use in this function)	spi.c	/grblHAL Driver STM32F756/Src	line 169	C/C++ Problem
'sdo' undeclared (first use in this function)	spi.c	/grblHAL Driver STM32F756/Src	line 168	C/C++ Problem
#error "Safety door cannot be enabled when > 3 motors are defined."	reference_map.h	/grblHAL Driver STM32H7xx/Inc	line 230	C/C++ Problem

Warnings

Description	Resource	Path	Location	Type
Invalid project path: Include path not found (GRBL Driver STM32F401/lwip/src/include).	GRBL Driver STM32F401		pathentry	Path Entry Problem
Invalid project path: Missing project folder or file: /grblHAL Driver STM32F756/grbl_esp32_if for Source path.	grblHAL Driver STM32F756		pathentry	Path Entry Problem
Invalid project path: Missing project folder or file: /grblHAL Driver STM32F756/oled for Source path.	grblHAL Driver STM32F756		pathentry	Path Entry Problem
Project 'grblHAL Driver STM32F756' has no explicit encoding set	grblHAL Driver STM32F756		/grblHAL Driver STM32F756	No explicit project encoding
Project 'STM32F030C8T6_Clock_Settings' has no explicit encoding set	STM32F030C8T6_Clock_Settings		/STM32F030C8T6_Clock_Settings	No explicit project encoding
Project 'STM32F030R8T6 Clock' has no explicit encoding set	STM32F030R8T6 Clock		/STM32F030R8T6 Clock	No explicit project encoding
Project 'STM32f401rct6 Clock Settings' has no explicit encoding set	STM32f401rct6 Clock Settings		/STM32f401rct6 Clock Settings	No explicit project encoding

Cheers and have a great day
George

dresco, 2024-03-09:
Hi,

Must admit I'm unclear what you are trying to achieve at first glance. Your logs contain a mixture of F0, F4, F7, and H7 build messages. From the following line, is still using the Nucleo 144 reference map, which doesn't tally with your description.

`#error "Safety door cannot be enabled when > 3 motors are defined." reference_map.h /grblHAL Driver STM32H7xx/Inc line 230 C/C++ Problem`

What's your starting point? Do you already have a supported dev board to build for and test on?

Cheers, Jon.

gxdeange, 2024-03-13:
Hi Jon,

Apologies for the above; here is what I have doen so far:

- I have created a custom design PCB that has the STM32H746ZIT6 MCU with custom pin mappings.
- I have used STM32CubeIDE to determine default pins for Ethernet, USB-OTG-HS, UART-8 and SPI-4
- I created 'MY_MACHINE_MAP' file and un-commented 'BOARD_MY_MACHINE'
- I changed the 'grbl/config' file from 3 Axis to 6 Axis
- I downloaded the plugins and core from terje io GitHub page
- I attempted to compile using 'Release H743 Nucleo 144

I am not sure why the logs are from different MCU's but I think that the code tried to use settings from 'reference_map.h' (as this is where the errors have been generated) instead of using 'my_machine_map.h' file.

Happy to zip and send any files (from STM32CubeIDE) that you think will be helpful

Cheers and have a great day
George

dresco, 2024-03-14:
Hi,

### MCU & peripheral support;

**MCU:**
Supports STM32H743 and STM32H723 parts. Both currently clocked at 480MHz. (You mention STM32H746 above, not sure if that is a typo?).

**Ethernet:**
Supports the same configuration & pinout as the 144 pin Nucleo boards. Specifically RMII connection to LAN8742 in REF_CLK Out Mode. (Have support for alternative pinout in another branch, but there are not many options for movement).

**USB:**
Supports USB FS only (PA11/PA12). Using 48MHz PLL clock source. No plans to support USB HS.

Note that the H723 is using HS peripheral in FS mode. (Is unusual amongst the H7 parts, in that it only has a HS peripheral, but is routed to the same pins as the FS peripheral on the other parts).

**UART/I2C/SPI:**
Supports multiple instances and pinouts, generally not an issue to add more.

**SDCard:**
Supported using SDMMC/SDIO only.

**SPI flash:**
Support for Winbond W25Q via QuadSPI is work in progress. (H723 will use OctoSPI peripheral in Quad mode).

**Clocks:**
The Nucleo boards have an 8MHz external clock, all other supported boards have 25Mhz external crystals. Some have external 32.768KHz crystals for RTC.

You mentioned HSI/LSI in an earlier post, this makes me wary, as generally not accurate enough for anything useful (even the cheapest Chinese dev boards come with external crystals).

Note that you will already need an accurate 25MHz clock source for the Ethernet PHY. Most cost effective would be 2x 25MHz crystals on the board, one for PHY and one for MCU, but could use a 25MHz oscillator and route to both?

I do plan to add support for HSI48 (with USB clock recovery), which will at least give a usable USB connection without a suitable external clock.

Regardless, if you want a different clock tree setup, would do this in main(). You'll see the conditionals for both MCU type and BOARD_ definition there..

### Building;

Check out the code and submodules together from this repo (will cause problems mixing versions).

Using my_machine is a good way to get started until you have a board name. 

I would suggest creating a new build configuration - `Properties, C/C++ Build, Create New Configuration, Copy settings from`... 
Then change details as required - `Properties, C/C++ General, Paths and Symbols`...
Remove the old BOARD_xxx setting and add your own etc. Probably best starting from the Nucleo 743 version if you want networking (and has the right linker files for H743ZI)..

Cheers, Jon.

closed 2024-05-25 by dresco