dresco/STM32H7xx #26: The CUBEIDE project does not have an. ioc configuration file

pzh11001, 2024-04-02:
Due to the lack of an. ioc configuration file. I am unable to visually understand the peripheral device configuration of the MCU through the STM32CUBEIDE graphical configuration tool. Can you add this file
Another issue: I am using version 1.15.0 of STM32CUBEIDE. After compiling the project, the "Build analyzer" in the bottom right corner cannot display any memory and FLASH usage information. I don't know if it's a bug caused by version updates

dresco, 2024-04-02:
Hi, can confirm I see the same Build Analyzer issue using v1.15 of STM32CubeIDE. I did jump all the way from 1.10 to 1.15, so am planning to try downgrading to 1.14 instead.

I don't actually have an .ioc file  for this project. Was removed early on, as the peripheral and clock setups vary between the different MCUs and boards anyway.  (I do tend to use throwaway CubeMX or CubeIDE projects to just verify hardware specific configurations where necessary)..

pzh11001, 2024-04-02:
in driver.h :

#if defined(BOARD_PROTONEER_3XX)
  #include "protoneer_3.xx_map.h"
#elif defined(BOARD_GENERIC_UNO)
  #include "uno_map.h"
#elif defined(BOARD_BTT_SKR_30)
  #include "btt_skr_v3.0_map.h"
#elif defined(BOARD_BTT_OCTOPUS_MAX)
  #include "btt_octopus_max_map.h"
#elif defined(BOARD_WEACT_MINI)
  //#include "weact_mini_map.h" //#####KKKKK Can only modify here
  #include "my_machine_map.h"
#elif defined(BOARD_REFERENCE)
  #include "reference_map.h"
#elif defined(BOARD_MY_MACHINE)
  #include "my_machine_map.h"   //#####KKKKK  Not working
#else // default board
  #include "generic_map.h"
#endif

dresco, 2024-04-02:
As you've noticed, each of the predefined build configurations define a number of settings within the project configuration. This is to allow them to build for each target without any code changes.

You will probably want to copy one as a starting point for your build, and change the symbols to suit your needs. 

Would be something like, Project -> Properties, Paths and Symbols, Manage Configurations, New... Give it a name and select an existing configuration to copy settings from (perhaps the WeAct H743 variant).

Then from that Paths and Symbols dialog, add a `BOARD_MY_MACHINE` symbol, and delete the `BOARD_WEACT_MINI` symbol. Building from that new configuration should then correctly reference your `my_machine_map.h` file

pzh11001, 2024-04-02:
Following the method you provided, I successfully compiled and set my own compilation options!
Thank you for your help

dresco, 2024-04-03:
> Following the method you provided, I successfully compiled and set my own compilation options! Thank you for your help

You're welcome.

For info, I downgraded my install to 1.14.x and the "Build Analyzer" window seems to be working again..

pzh11001, 2024-04-03:
It seems that the problem is indeed caused by a BUG in the CUBEIDE version

closed 2024-05-25 by dresco