dresco/STM32H7xx #28: Octopus Max no com port

T4KUUY4, 2024-05-16:
I have some issues with the compiled binary for an OCtopus MAX ez, H723 with 5160 and sense resistor set to 50. MPG mode is also set to TEST2. The firmware flashes normally over DFU, but doesnt show up in device manager nor lsusb on linux. Only after 20 seconds as "unknown device descriptor". Is there a fix for this?

dresco, 2024-05-17:
Hi, possibly a bug in the spindle PWM settings that I haven't had a chance to look into yet.

See [this](https://github.com/dresco/STM32H7xx/issues/18#issuecomment-2083360445) comment, and maybe try building with spindle set to on/off to test?

T4KUUY4, 2024-05-17:
> Hi, possibly a bug in the spindle PWM settings that I haven't had a chance to look into yet.
> 
> See [this](https://github.com/dresco/STM32H7xx/issues/18#issuecomment-2083360445) comment, and maybe try building with spindle set to on/off to test?

Thanks for the quick Response. I havent tried that yet. Ill compile with spindle disabled and let you know.

Thanks,
Tom

dresco, 2024-05-17:
@T4KUUY4  Have just taken a quick look at the code, and see what's wrong. PA1 isn't defined as an available PWM port, I suspect is just crashing during startup.

Will fix in the next update, hopefully next week..

Cheers, Jon.

T4KUUY4, 2024-05-17:
> @T4KUUY4  Have just taken a quick look at the code, and see what's wrong. PA1 isn't defined as an available PWM port, I suspect is just crashing during startup.
> 
> Will fix in the next update, hopefully next week..
> 
> Cheers, Jon.
> 

Oh nice find!
Thanks for the update and for looking at it. 
I can try to set that pin manually today+ flash to see if it works.

T4KUUY4, 2024-05-17:
Yep. Can confirm, setting the spindle to on/off, makes the board show up with a com port

T4KUUY4, 2024-05-17:
One more question regarding the output pins like fans etc. 
How can i enable these to say... control a fan via a gcode command?

dresco, 2024-05-17:
> How can i enable these to say... control a fan via a gcode command?

Yes, I think either by using the fans plugin, or I believe there are M codes for controlling unclaimed aux outputs? I've not used either tbh, so may be better off asking in the core repo..

I'll check the web builder configuration, the digital outputs probably just need to be defined there also.

Depending what your plans are for a spindle, you may need additional outputs defining in the map file as well, as there are currently just the three aux outputs that may be mapped to spindle control.

T4KUUY4, 2024-05-17:
> Yes, I think either by using the fans plugin, or I believe there are M codes for controlling unclaimed aux outputs? I've not used either tbh, so may be better off asking in the core repo..
>
> I'll [...]

I added 4 more outputs via the board map file and these work perfectly. the only thing is, the driver seems to freak out at 3+ fans enabled. if fans_enable is set to 2, i can control these with M106/7 P0 and P1

`//FAN PORTS
#define AUXOUTPUT3_PORT             GPIOA   // FAN2
#define AUXOUTPUT3_PIN              4
#define AUXOUTPUT4_PORT             GPIOA   // FAN3
#define AUXOUTPUT4_PIN              3
#define AUXOUTPUT5_PORT             GPIOF   // HE3
#define AUXOUTPUT5_PIN              7
#define AUXOUTPUT6_PORT             GPIOF   // HE2
#define AUXOUTPUT6_PIN              9
`

T4KUUY4, 2024-05-18:
With "the driver seems to freak out", I mean that only 2 fans  show up (if 4 selected. 4 outputs are available as per above). But not as Fan0 and Fan1. more like Fan2 and Fan3. so 2 are missing and also cant be assigned with $386 etc

T4KUUY4, 2024-05-20:
> > With "the driver seems to freak out", I mean that only 2 fans show up but they dont work
> 
> @T4KUUY4
> 
> I'll take a look, am working on a board here with 4 aux outputs..
> Cheers, Jon.

Awesome. Thank you very much.

dresco, 2024-05-21:
Have played around a bit with the fans plugin here, and seems to be working as expected for me?

It did take me a while to work out that I needed to explicitly map each fan to an available Aux input with the $386-$389 settings, before that I was getting an error on startup that a configured port number wasn't available.

I have some other changes queued up to push, so I'll address the PWM spindle output next.

T4KUUY4, 2024-05-21:
> Have played around a bit with the fans plugin here, and seems to be working as expected for me?
>
> It did take me a while to work out that I needed to explicitly map each fan to an available Aux input [...]

Interesting. For me, i couldnt even map it because it said $386 isnt an available command or so. I can have a look at it aswell

dresco, 2024-05-21:
> For me, i couldnt even map it because it said $386 isnt an available command or so.

Yes, I also noticed that the settings code in the fan plugin worked backwards through the array of fan outputs, and didn't get any further than the one it failed on.

So initially I only had setting $389 (Fan3) available (which was failing, as it conflicted with an Aux port being used for spindle control), but after setting that to another free port & resetting, it carried on with $388 (Fan2), $387 (Fan1), and $386 (Fan0).

dresco, 2024-05-21:
Could you test the following patch for me please? Should fix the PWM spindle issue for the Octopus Max..

```
diff --git a/Src/pwm.c b/Src/pwm.c
index ebc9b60..67821de 100644
--- a/Src/pwm.c
+++ b/Src/pwm.c
@@ -38,6 +38,10 @@ static const pwm_signal_t pwm_pin[] = {
         .en = timerCCEN(1, N), .pol = timerCCP(1, N), .ois = timerCR2OIS(1, N), .ocm = timerOCM(1, 1), .ocmc = timerOCM(1, 1)
     },
 #endif
+    {
+        .port = GPIOA, .pin = 1, .timer = timer(2), .ccr = &timerCCR(2, 2), .ccmr = &timerCCMR(2, 1), .af = timerAF(2, 1),
+        .en = timerCCEN(2, ), .pol = timerCCP(2, ), .ois = timerCR2OIS(2, ), .ocm = timerOCM(1, 2), .ocmc = timerOCM(1, 2)
+    },
     {
         .port = GPIOA, .pin = 8, .timer = timer(1), .ccr = &timerCCR(1, 1), .ccmr = &timerCCMR(1, 1), .af = timerAF(1, 1),
         .en = timerCCEN(1, ), .pol = timerCCP(1, ), .ois = timerCR2OIS(1, ), .ocm = timerOCM(1, 1), .ocmc = timerOCM(1, 1)

```

T4KUUY4, 2024-05-21:
Sure thing. Let me grab my board

T4KUUY4, 2024-05-21:
Okay so the board flashed fine and is now detected. i only have 5V available right now through USB so i cant test my 0-10V spindle pwm voltage right now. but the board is showing up fine now.

T4KUUY4, 2024-05-21:
between which 2 pins should i see 5V (assuming i set it to 12k, on a 24k max spindle). V+ and PWM?

dresco, 2024-05-21:
Thanks for testing so quickly.

You should see the actual PWM signal on the FAN4 header. Am not sure any of the BTT boards have a native 0-10V analog output do they? Will check the published schematic tomorrow..

T4KUUY4, 2024-05-21:
> Thanks for testing so quickly.
> 
> You should see the actual PWM signal on the FAN4 header. Am not sure any of the BTT boards have a native 0-10V analog output do they? Will check the published schematic tomorrow..

Glad i could help. Ss far as im aware, they dont have a native 0-10V analog output. They only have PWM outputs with selectable Voltage on the V+ pin. (5, 12, 24). I have no problem with using a PWM to analog converter if thats possible to use.

dresco, 2024-05-22:
> They only have PWM outputs with selectable Voltage on the V+ pin. (5, 12, 24). I have no problem with using a PWM to analog converter if thats possible

Yes, looks that way to me also. Another option could be modbus control of the VFD (if supported). There is power + serial port on the TFT header in the bottom right for this - would just need a cheapo external RS485 transceiver board.

T4KUUY4, 2024-05-22:
Thats true. I need the uart for a smartPendant so i cant use modbus sadly. I will go for the pwm-analog converter. As soon as my drivers are here, ill install everything to fully test.

dresco, 2024-05-25:
Spindle PWM fix is now available in the [Web Builder](http://svn.io-engineering.com:8080/)

closed 2024-05-25 by dresco