dresco/STM32H7xx #29: Octopus max 5160ez no motors

T4KUUY4, 2024-06-15:
Hi and sorry for creating another issue.
I just cant get my head around why the motors are not working. Ive got 5160 ez on my octopus max. In the First 4 Slots. 

Ive compiled on the webbuilder with ganged Y motors and 5160s with 0.075 sense resisor. Ive reset my grblhal settings but i still get no motor movement. I tried inverting the enable pins but no change.

Also, even though i havent fully confirmed yet, enabling mpg test 2 in the Firmware, makes it Crash a lot. Mostly "controller is halted" or error:20. 

Any ideas?

Thanks in advance 
Tom

T4KUUY4, 2024-06-15:
My MPG wasnt connected yet as the steppers didnt even move.
That might be why the firmware crashes with MPG enabled. but im not 100% sure

T4KUUY4, 2024-06-15:
M122 returns "could not communicate with stepper driver"

T4KUUY4, 2024-06-15:
Update. They are detected when running Klipper. so the drivers, are fine

T4KUUY4, 2024-06-15:
I feel a little... stupid ngl.
one stepstick got considerably hotter than the rest. so i removed that one, and the other 3 showed up ...

T4KUUY4, 2024-06-15:
MPG works aswell when its connected.

dresco, 2024-06-16:
Glad you got it working in the end.

I don't know if there is anything different with the driver configuration vs Klipper/Marlin? I do know that the stepper max current (from $14x parameters) gets set before enabling the drivers, as that was something I checked previously.

I've got an SKR3 here with both 2209 and 5160 EZ drivers, and haven't experienced any issues with them. That said, it's only been used for development, running Nema17 motors with no load on them.

T4KUUY4, 2024-06-16:
> Glad you got it working in the end.
>
> I don't know if there is anything different with the driver configuration vs Klipper/Marlin? I do know that the stepper max current (from $14x parameters) gets [...]

Yea. sorry for all the notifications :D Turns out 2 were actually dead and not only one. grbl wont initialize the other ones if a dead one is connected.

dresco, 2024-06-26:
Hi @terjeio - the above issue has come up for a few people now, needing a delay before the Trinamic drivers are initialised or else getting a communication error. I've not been able to replicate, but perhaps dependent on rise time for the modules power supply?

Anyway, do you have an opinion on adding a (configurable?) startup delay directly to the Trinamic drivers? I could add a delay to the H7 code before calling trinamic_init(), but I'd expect it might crop up in other drivers also?

dresco, 2024-06-27:
@T4KUUY4 

Just to check something - I've been assuming this issue only occurs on power-up, is that the case? 
i.e. if the board is already powered, and is restarted with the reset button then it's okay?

T4KUUY4, 2024-06-27:
@dresco correct. Only on power up

terjeio, 2024-06-27:
Start with increasing the delay [here](https://github.com/grblHAL/Plugins_motor/blob/87b9056327a87e7a7784a4c9addfb046050199f7/trinamic.c#L2415) - to 500 (ms) initially?. @T4KUUY4 are you compiling locally or using the Web Builder? If using the Web Builder I can modify the code so that you can rebuild and test.

T4KUUY4, 2024-06-27:
> Start with increasing the delay [here](https://github.com/grblHAL/Plugins_motor/blob/87b9056327a87e7a7784a4c9addfb046050199f7/trinamic.c#L2415) - to 500 (ms) initially?. @T4KUUY4 are you compiling [...]

Hi,
Im using the webbuilder right now. 
Thanks a bunch

T4KUUY4, 2024-07-08:
@terjeio @dresco That seems to do the trick. But the PROBE input isnt working on the Octo max ez. I compiled via webbuilder and locally and it just says "NOPROBE" in NEWOPT. Any ideas?

dresco, 2024-07-08:
> @terjeio @dresco That seems to do the trick. But the PROBE input isnt working on the Octo max ez. I compiled via webbuilder and locally and it just says "NOPROBE" in NEWOPT. Any ideas?

No, but was also mentioned in another issue [here](https://github.com/grblHAL/core/discussions/203#discussioncomment-9979643). Will check what's going on, might take me a day or two though..

T4KUUY4, 2024-07-08:
@dresco Sure, give me a few minutes and ill report back. Im at the CNC atm anywasy

T4KUUY4, 2024-07-08:
@dresco seems like it doesnt like that at all. 
`<command-line>:0:0: note: this is the location of the previous definition
In file included from Src\driver_spindles.c:30:0:
./grbl/pin_bits_masks.h:27:2: error: #error "Probe input is not supported in this configuration!"
 #error "Probe input is not supported in this configuration!"`

T4KUUY4, 2024-07-08:
im pulling a new git down now. just to verify

dresco, 2024-07-08:
Hmm, that implies `PROBE_ENABLE` is defined, but `PROBE_PIN` is not. Does that section of your map file look like this?

[image]

T4KUUY4, 2024-07-08:
i might have misinterpreted your code sample above. i thought that id need to remove the previous probe_pin and port (because of the - in front.)  Compiling with the new code now.

dresco, 2024-07-08:
> i might have misinterpreted your code sample above. i thought that id need to remove the previous probe_pin and port (because of the - in front.) Compiling with the new code now.

Yeah sorry, that diff wasn't perhaps the easiest to parse visually :)
Should just need to add the new AUXINPUT1_PORT/PIN lines, and change the PROBE_PORT/PIN lines..

T4KUUY4, 2024-07-08:
yep. compiles now. let me run down again to flash.

T4KUUY4, 2024-07-08:
@dresco that did the trick. Works like a charm now

dresco, 2024-07-08:
@T4KUUY4 Cool, thanks for the confirmation. Will get the changes pushed up to GitHub later..

T4KUUY4, 2024-07-22:
@dresco Awesome Thanks.
On another note.. Have you ever had an issue with motors not moving if you enable the "do not turn off motor in idle" thing? I dont know the exact $ setting atm but i can look later. 

Basically, if you set the motors to not disable in idle (so they cant move), you cant jog/move anymore. Z Homing after a reset seems to work, but nothing will move after that. I dont know if thats a bug on my end or a general thing. 

Also, is the probe input pulled up? i seem to notice that it sometimes flickers as if it was floating for a second. Nothing gets interrupted though so it might be a visual bug.

T4KUUY4, 2024-07-22:
> Not that I'm aware of, my CNC router has always been using `$1=255` to keep the steppers enabled. I don't have Trinamic drivers on any real machines, but I can dig out my SKR3 with the EZ5160 modules and double check on the bench..
> 
That would be awesome. let me know if you need anything tested. Thanks

 
> Seems to be, the Octopus MAX schematic shows a 10K pullup to 3.3V
Good to know. Must be a visual bug then.

dresco, 2024-07-22:
Afraid I'm not able to replicate here.

My test setup is the standard SKR3 build from platformio.ini (5160, 3 axis, bootloader), with 3x steppers on X/Y/Z outputs.

Reset all parameters, then `$1=255`, `$5=7`, `$14=70`, & `$22=3`. Can move all motors before and after Z homing as expected. (I faked Z homing just shorting the input to ground during the process).

T4KUUY4, 2024-07-22:
Thanks for testing it. Then its either a buggy configured firmware on my end. Or because im using a ganged Y axis.
Im building a second CNC atm with only 1 y axis motor so ill give it a shot there aswell. Same controller and stuff.

terjeio, 2024-07-22:
What is your $4 setting (Invert stepper enable pins)?
FYI $1 (Step idle delay) can be set <> 255 to disable steppers when idle, except for axes configured with $37.

dresco, 2024-07-22:
> What is your $4 setting (Invert stepper enable pins)?

$4=7

> FYI $1 (Step idle delay) can be set <> 255 to disable steppers when idle, except for axes configured with $37.

Interesting, thanks. Hadn't noticed that $37 setting before..

T4KUUY4, 2024-08-24:
@dresco @terjeio So ive did some more testing today on a different CNC project and i found that having stealthchop turned on by default in the motor_plugin tmc5160 section, made a drastic difference in terms of performance. 

I have some stepperonline 23he45-4204s motors here that i never managed to get over a certain speed (1300mm/min in my case). Even with nothing attached. (32 microsteps, 2A of current).  It for some reason worked with some load on the X motor, but upon taking it out, it didnt again. which was very weird. I initially thought of having a dead motor so i just swapped in a replacment. issue was still present. 

Thats when i noticed that stealthchop, is hardcoded on. Why is that a thing?
Can that be toggled in IoSender for example? Or a dropdown selection in the webbuilder?

dresco, 2025-06-10:
Hi @T4KUUY4, closing due to inactivity. 

If stealthchop is still an issue, can you open directly in the [Plugins_motor](https://github.com/grblHAL/Plugins_motor) repo? Thanks!

closed 2025-06-10 by dresco