dresco/STM32H7xx #32: Not status reports when sending "?" from a self built Xojo application

richardklingler, 2024-10-04:
Strange issue with the Octopus Max board and status reports using "?".

When using the macOS app "Serial" and just typing "?" I get immediately the status back.
Sending "?" from a Xojo compiled macOS app nothing is returned though "?" is sent out as checked with two FTDI serial adapters back2back.

Tried with a fluidNC board and a ATMEGA2560 board, both return the status when sending "?" from the self built Xojo application.

Just reflashed the STM32H723 with a firmware from the web builder...still no status report is coming back.

Needs there something to be setup in advance to get status reports?
When sending "$$" I get a response back...

dresco, 2024-10-04:
Hi, 

> Needs there something to be setup in advance to get status reports?

Nothing I can think of, "?" is one of the special [realtime](https://github.com/gnea/grbl/blob/master/doc/markdown/commands.md#grbl-v11-realtime-commands) commands that should be interpreted immediately..

> When sending "$$" I get a response back...

How odd! Do you have any code I could try here? My dev environment is Linux, but my wife has a MacBook if needed..

richardklingler, 2024-10-04:
Tried on Linux as well...behaves exactly the same and I can't see inside USB CDC what is happening (o;

Here is the Xojo project....modified it a little as it can be compiled and run on Linux as well without any license....
You can get Xojo at https://www.xojo.com

[Rapi.xojo_binary_project.zip](https://github.com/user-attachments/files/17258717/Rapi.xojo_binary_project.zip)

dresco, 2024-10-04:
Thanks, found the issue I think.

DTR wasn't being asserted on the USB-CDC connection from the Xojo app, and was therefore failing a [connection check](https://github.com/dresco/STM32H7xx/blob/fe859dce9cb3c9a713c1eb5e0a48e1e798b5a4ab/Src/usb_serial.c#L42-L45) before sending the status response. Not sure why that doesn't affect more output, but hey ho..

Looking at your code, I tried setting `SerialConnection1.DTR = True`, but no change. Then google [led me to ](https://forum.xojo.com/t/serial-dtr-true/78556/15)`SerialConnection1.DataTerminalReady = True`, and that seems to have done the trick here.

richardklingler, 2024-10-04:
Hey...that's just awesome :-)

Many thanks (o;

I'm sure I come back as I might need your help to port some parts that are not included in the H723 port like THC...

closed 2024-10-04 by richardklingler