dresco/STM32H7xx #34: Y_GANGED/AUTOSQUARE mode

robertcigan, 2024-11-09:
Hi, this is my setup
BTT SKR 3.0 EZ H723 with 4 motors connected on TMC5160 

With default mode 1 motor per axis XYZ it works. 
When enabled, Y_GANGED/AUTOSQUARE, Y motor on Y driver does not work (4th driver with Y2 motor works). I tested switching Y driver to B driver (5th position) to see if it's remapped or so but it's not.

The only options that I enabled are Y_GANGED and Y_AUTO_SQUARE, but maybe I'm doing something wrong. Has anyone tested ganged axis?

Thanks a lot for help of any kind.

robertcigan, 2024-11-09:
[20:20:01] [ $pins ] [PIN:PA5,Feed hold]
[20:20:01] [ $pins ] [PIN:PA6,Cycle start]
[20:20:01] [ $pins ] [PIN:PA4,Emergency stop]
[20:20:01] [ $pins ] [PIN:PC13,Probe]
[20:20:01] [ $pins ] [PIN:PC1,X limit min]
[20:20:01] [ $pins ] [PIN:PC3,Y limit min]
[20:20:01] [ $pins ] [PIN:PC2,Y limit min 2]
[20:20:01] [ $pins ] [PIN:PC0,Z limit min]
[20:20:01] [ $pins ] [PIN:PA7,Aux in 0,P0]
[20:20:01] [ $pins ] [PIN:PD4,X step]
[20:20:01] [ $pins ] [PIN:PA15,Y step]
[20:20:01] [ $pins ] [PIN:PD15,Y2 step]
[20:20:01] [ $pins ] [PIN:PE2,Z step]
[20:20:01] [ $pins ] [PIN:PD3,X dir]
[20:20:01] [ $pins ] [PIN:PA8,Y dir]
[20:20:01] [ $pins ] [PIN:PD14,Y2 dir]
[20:20:01] [ $pins ] [PIN:PE3,Z dir]
[20:20:01] [ $pins ] [PIN:PD5,Motor CSX]
[20:20:01] [ $pins ] [PIN:PD0,Motor CSY]
[20:20:01] [ $pins ] [PIN:PE1,Motor CSZ]
[20:20:01] [ $pins ] [PIN:PC6,Motor CSM3]
[20:20:01] [ $pins ] [PIN:PD6,X enable]
[20:20:01] [ $pins ] [PIN:PC7,Y enable]
[20:20:01] [ $pins ] [PIN:PD1,Y enable]
[20:20:01] [ $pins ] [PIN:PE0,Z enable]
[20:20:01] [ $pins ] [PIN:PB6,Spindle on]
[20:20:01] [ $pins ] [PIN:PB5,Spindle direction]
[20:20:01] [ $pins ] [PIN:PB0,Spindle PWM]
[20:20:01] [ $pins ] [PIN:PB4,Mist]
[20:20:01] [ $pins ] [PIN:PB3,Flood]
[20:20:01] [ $pins ] [PIN:PE15,MISO,Motor]
[20:20:01] [ $pins ] [PIN:PE13,MOSI,Motor]
[20:20:01] [ $pins ] [PIN:PE14,I2C SCK,Motor]
[20:20:01] [ $pins ] [PIN:PA10,RX,UART1]
[20:20:01] [ $pins ] [PIN:PA9,TX,UART1]

dresco, 2024-11-09:
Hi, yes ganged axis should work. Will try to replicate on my SKR3 tomorrow. 

Is it a local build or from the web builder?

robertcigan, 2024-11-09:
Local in Code/Platformio changing -D params in platformio.ini

dresco, 2024-11-09:
> Local in Code/Platformio changing -D params in platformio.ini

Thanks, was going to ask for your binary to try, but then remembered my SKR3 is the original H743 variant, so that won't help..

Have just quickly tried on my SKR3 here and seems to be working as expected (ganged Y output on YM & E0M outputs). My only change was to add `-D Y_AUTO_SQUARE=1` to the platformio.ini section for this board.

[image]

robertcigan, 2024-11-11:
I spend few hours testing yesteday and for some reason at some point it started working after I removed all 5160 drivers and installed them again. It was always working weird in any of the settings except default so it might have been a HW issue with a connection or something. I wanted to test the web builder, but for some reason with default settings whenever I selected ganged or auto square option, the web build failed. Anyway I have no conclusion what was wrong, but it's working now, thanks for your feedback. I kinda regret going the BTT SKR 3 way, but I like this compact format so much, because I don't want to have a big box of electronics with those excessively large drivers.

closed 2024-11-11 by robertcigan

dresco, 2024-11-11:
> I wanted to test the web builder, but for some reason with default settings whenever I selected ganged or auto square option, the web build failed.

Thanks for reporting, will look into that. Web Builder is also failing for me with Trinamic drivers selected.