dresco/STM32H7xx #35: Grblhal for Bigtreetech Manta m8p v2

Sillister, 2024-11-16:
Hello Mr./Mrs. dresco,

I'm relatively new to the firmware scheme and would like to know if I can create/ modify Grblhal for my Bigtreetech Manta m8p v2 board?

Are there any guides I can follow, or would you like to help me?

What tools are required?

If testing is required, I can provide it with a 3 axis cnc

Thanks in advance

dresco, 2024-11-17:
Hi, 

It's not a board I'm familiar with, but looking at the [schematic](https://github.com/bigtreetech/Manta-M8P/blob/master/V2.0/Hardware/BIGTREETECH%20MANTA%20M8P%20V2.0-SCH.pdf), I don't think it will be too hard to add basic support.

What does your current configuration look like in terms of motor drivers, limit switches etc?

Cheers,
Jon.

Sillister, 2024-11-20:
Hello Jon,

first of all, sry for the slow reply.

That sounds great, I will send you my configuration today or tomorrow, i dont have access to my cnc right now.

I have 2 small questions, What kind of features wouldn't work with basic support, something major?

Is it possible to configure 2 limit switches on one axis?

Cheers,
Manuel

Sillister, 2024-11-20:
[image]
[image]

So, here are my connections and the diagram.

I have connected Motor M2 (DIR: PE1, STEP: PE2, EN: PE4) for the x - axis,

M3 (DIR: PB7, STEP: PB8, EN: PE0) for the z- axis, 

M7 (DIR: PD3, STEP: PD4, EN: PD6) and M8 (DIR: PC8, STEP: PC7, EN: PD2) for the y - axis.

I have 5 limit switches (the picture is older, I added 1):
X: PF4
Y: PF3
Z1: PF1
Z2: PF0
external Z to level with equipped tool: PC15

I have 3 z sensors because of the safety, I have a 2,2kw spindle with a weigt of about 4kg. So i wanted to make sure it doesnt do anything insecure. One top z limit, one bottom and 1 to level z with equipped tool.

That should be all, if I'm not mistaken.

dresco, 2024-11-20:
Thanks,

> What kind of features wouldn't work with basic support, something major?

I was just looking at all the connections/features on the board. Will start with the essentials (motor outputs, limit switches, control inputs etc) and go from there...

> Is it possible to configure 2 limit switches on one axis?

As far as I'm aware, just one limit input per axis. Multiple switches would be then wired in series if NC (preferred), or in parallel if NO.

Any reason for not using motors 1/2/3 for x/y/z? I'd prefer to use these for the default map, less chance of surprises for those who follow.. ;)

Your 3rd limit input on the Z axis sounds like it would be the probe input (PD13)?

Also, will need to know inputs/outputs you're using for the controls (cycle start, feed hold, reset), and for spindle control?

Sillister, 2024-11-20:
Ok, then I will need to rewire the switches into series thanks.

The reasok I didn't connect 1,2,3,4 was because of cable Management: 
[image]
I would prefer not to rewire these because they are shitty to crimp. I already did a dunctiin test with klipper, the connections work fine.

You are right I need to wire it to the probe pin not endstop, I thought I could use another endstop pin as probe, but that wont be a problem, I can rewire that.

And for your last question, right now I dont control the spindle with the board. I want to add it in the future, but I need to get familiar with the schematic of the spindle driver (Sourcetronic ST500 0.75KW - 7.5KW 230V), if you have a quick advice on what to connect where on the board I will add it.

Thank you for your time

dresco, 2024-11-20:
> I would prefer not to rewire these

That's cool, if you building locally you can just cut & paste the motor definitions around, or create your own my_machine_map.h with the custom motor ordering.

I'll create a new branch with a board definition / map file for the Manta as a starting point, are you comfortable compiling with PlatformIO?

For spindle control, the options would be;
(a) enable & direction pins, with either a PWM pin, or a 0-10V analog output for speed control. (Analog output will require external circuitry if not already on the board).
(b) Modbus control (requires external RS485 transceiver if not already on the board). Doesn't appear to have RS485 onboard, but I'll map a UART to the TFT header.

Give me a day or two and I'll add the support to a new branch..

Cheers, Jon

Sillister, 2024-11-20:
Thank you very much, I will look into it and send updates.

I program Arduino and Esp with PlatformIO, if its the same, then yeah i can compile it.
I wanted to learn how to create this firmware aswell, so I see it as a challenge

Regaeds, Manuel

dresco, 2024-12-01:
Hi Manuel,

I've created a [branch](https://github.com/dresco/STM32H7xx/tree/manta) with the initial support for the Manta board. Please try it and let me know how you get on.

Info regarding pin mappings will be in the [board map](https://github.com/dresco/STM32H7xx/blob/manta/boards/btt_manta_m8p_v2_map.h) file. The schematic can be found in the BTT repository.

A few things to note;

- There are only three opto-isolated inputs on the board. I've used these for the mandatory cycle start, feed hold, and reset/halt inputs.
- The board is configured for maximum 4-axis, due to interrupt clashes between two of the six limit pin inputs and the above control inputs.
- The input assigned for the probe has no protection on it, is wired directly to the MCU pin.
- SDcard won't currently work, as only SDMMC connections are supported in the H7 driver, and this board is wired with just SPI connection to the card slot.
- I2C on the board header won't currently work, as the driver expects both pins to be on the same port, which is not the case here.

Appreciate that you will want to change the motor / limit switch assignments to match your existing wiring, hopefully that will be clear enough within the map file..

Regards,
Jon.

SergeTauger, 2025-06-05:
> I've created a [branch](https://github.com/dresco/STM32H7xx/tree/manta) with the initial support for the Manta board.

Hi. Is manta m8p v2 moved to master branch now, or we should take the one from dedicated manta branch?

SergeTauger, 2025-06-10:
I agree, but at the time these are only I have at hand :) should I do some fixup to merge with main or just suggest this branch abandoned and use as is?