dresco/STM32H7xx #36: Issue with releasing a stepper motor spindle after claim

Freezerohmatic, 2025-01-31:
I found a bug in the driver.c file in `stepperClaimMotor`. 

When a stepper motor is claimed as a spindle it may not be fully released by this function because the coinciding `step_pulse.inject.axes.bits` may be set and it will not be cleared after the stepper is release. Other repos by grblhal (I'm assuming terjeio) add `step_pulse.inject.axes.bits = step_pulse.inject.claimed.bits;` after `step_pulse.inject.claimed.mask &= ~(1 << axis_id);` to clear any non claimed bits.

so 

`
else
        step_pulse.inject.claimed.mask &= ~(1 << axis_id);
`

becomes 

`
else{
    step_pulse.inject.claimed.mask &= ~(1 << axis_id);
    step_pulse.inject.axes.bits = step_pulse.inject.claimed.bits; 
}
`

dresco, 2025-01-31:
Thanks, lets check what the intention is with @terjeio
I see the above line in the [ESP32](https://github.com/grblHAL/ESP32/blob/f14535426777b5b6a2f274e365461fb9eae4f696/main/driver.c#L1339C1-L1347C2) & [RP2040](https://github.com/grblHAL/RP2040/blob/d08efbf8d6581a56a3acb9150cf6433390ec0bb5/driver.c#L1284-L1292) drivers, but not the others (F4, F7 etc)?

Freezerohmatic, 2025-02-01:
Interesting, I was playing with a FluidNC (ESP32) board before jumping to a stm32h7 version. (I couldn't easily get stepper spindle working using I2S streaming, I did get something working but the interrupts varied so much it honestly wasn't safe).

terjeio, 2025-02-01:
Fix [committed](https://github.com/grblHAL/STM32F4xx/commit/c85a5000495211d07a0faf1a496614a7c6103506#diff-78adbf68c4518084f13b26f19ead59c48808a6f2a6a1823a062b91223af556f8).

There is another piece in this puzzle, the motor is only available for axis commands if it is [permanently enabled](https://github.com/grblHAL/Plugins_spindle/blob/54ab2ab84c2c0770e1436fc13efae88138451e60/stepper.c#L57-L58) with `$37`.

The RP2040 is a bit different since outputting step pulses does not require a second interrupt to turn it off.

Freezerohmatic, 2025-02-02:
The permanently enabled point confuses me. See https://github.com/grblHAL/Plugins_spindle/issues/37

From my testing I cannot see any issues caused by making this change but I don't know your code base well enough to state with certainty.  It may be fine in this repo and not others?

dresco, 2025-02-05:
> Fix [committed](https://github.com/grblHAL/STM32F4xx/commit/c85a5000495211d07a0faf1a496614a7c6103506#diff-78adbf68c4518084f13b26f19ead59c48808a6f2a6a1823a062b91223af556f8).

Thanks, fix now also committed to H7 repo.

closed 2025-06-10 by dresco