dresco/STM32H7xx #37: Proper wiring of limit switch for Y2 (E1) in SRK 3 EZ

davidqmora, 2025-02-02:
Hi. Not sure where else to ask this. If this is the wrong forum, I'd appreciate any pointers to the right one.  

I have successfully built and deployed the firmware for the SKR 3 EZ board with TMC2209 drivers. It works pretty well, but I am having a problem when homing the Y access. Here are the facts:

 - Built with Platformio (HEAD commit).
 - X_GANGED=1, X_AUTO_SQUARE=1, Y_GANGED=1, Y_AUTO_SQURED=1
 - Limit switches: 
   - Two for X: GPIOC1, GPIOC2
   - Two for Y: GPIOC3, GPIOC15 (PWRDET)
   - One for Z: GPIOC0
 - All the switches are detected for all axes.
 - Homing sequence set to Z first, X second, Y third.

The homing works fine for Z and X, but for Y, the switch connected to PWRDET even though it clicks (and is detected), does not cause the corresponding motor to stop. The net effect is one side of Y stopping at the switch and the other one just skipping as it continues pushing hard against the stop. After some head scratching and revisiting boards\btt_skr_v3.0_map.h, I re-read this comment:

```
// If using TMC drivers, jumper from PWRDET connector pin 3 to DIAG pin on driver.
```

So, I am guessing that my problem is related to this particular connection (I have not tied PWRDET to DIAG in M4). My question is, being that DIAG is an output in the TMC2209, am I supposed to only connect that pin to PWRDET and not use a limit switch, or am I supposed to connect both the switch and DIAG to PWRDET? At the moment, PRDWT is connected to an NC switch to ground, so I am a bit concerned about contention.

Thanks.
David Mora

dresco, 2025-02-03:
Hi, yes that comment came from the SKR2 map file (F4xx driver), and I believe is only required for sensorless homing - not for use with a limit switch.

I've tried replicating your issue on the bench here, and appears to be working as expected for me.

I've got steppers on Y & Y2, and both limit inputs jumpered to ground (pullups are enabled as default). Checking the switch status with `$HSS` appears as expected.

Am starting single axis homing with `$HY` and whichever jumper wire I disconnect first - that motor stops immediately. The other stops after another second or so (when the max offset is reached). Note I haven't tried homing more than a single axis, as am unable recreate the timing for a successful homing/pull-off sequence by hand..

[image]

These were the only build options I added;

```
  -D Y_AUTO_SQUARE=1
  -D X_AUTO_SQUARE=1
```

And my `$pins` output looks like this;
```
[PIN:PA5,Feed hold]
[PIN:PA6,Cycle start]
[PIN:PA4,Emergency stop]
[PIN:PC13,Probe]
[PIN:PC1,X limit min]
[PIN:PC2,X limit min 2]
[PIN:PC3,Y limit min]
[PIN:PC15,Y limit min 2]
[PIN:PC0,Z limit min]
[PIN:PA7,Aux in 0,P0]
[PIN:PD4,X step]
[PIN:PD15,X2 step]
[PIN:PA15,Y step]
[PIN:PD11,Y2 step]
[PIN:PE2,Z step]
[PIN:PD3,X dir]
[PIN:PD14,X2 dir]
[PIN:PA8,Y dir]
[PIN:PD10,Y2 dir]
[PIN:PE3,Z dir]
[PIN:PD5,Motor CSX]
[PIN:PD0,Motor CSY]
[PIN:PE1,Motor CSZ]
[PIN:PC6,Motor CSM3]
[PIN:PD12,Motor CSM4]
[PIN:PD6,X enable]
[PIN:PC7,X enable]
[PIN:PD1,Y enable]
[PIN:PD13,Y enable]
[PIN:PE0,Z enable]
[PIN:PB6,Spindle on]
[PIN:PB5,Spindle direction]
[PIN:PB0,Spindle PWM]
[PIN:PB4,Mist]
[PIN:PB3,Flood]
[PIN:PE15,MISO,Motor]
[PIN:PE13,MOSI,Motor]
[PIN:PE14,I2C SCK,Motor]
[PIN:PA10,RX,UART1]
[PIN:PA9,TX,UART1]
```

Just one other thought, your Y & Y2 switches and motors are not mixed up? I've had similar symptoms myself when I accidentally switched up the sides for the limit switch and motor wiring, so the wrong motor was stopped & the other just continued trying to push though the end of travel..

Cheers, Jon.

davidqmora, 2025-02-03:
Thank you for the quick response. Your last comment about the mixed switches and motors immediately made sense. I feel silly for not thinking about it. The switches were fine, but I must have mistakenly swapped the motor control wires at some point. It all works fine now that everything has been wired correctly.

closed 2025-02-03 by davidqmora