dresco/STM32H7xx #38: Is this a new feature? Send "?" only reply with ok

hanke-cnc, 2025-02-12:
Sending "?" to it only replies with "ok." However, when using ioSender, you can receive a response. ioSender probably uses another real-time command to obtain the reply.

[image]

[image]

[image]

hanke-cnc, 2025-02-12:
I haven't updated the version for a long time. I don't know if this is a bug or a new feature.

dresco, 2025-02-12:
Terminal program not sending the expected character maybe? I would check it's really sending an ASCII '?' (63 or 0x3F).

The "ok" will come from the new line.

hanke-cnc, 2025-02-12:
> Terminal program not sending the expected character maybe? I would check it's really sending an ASCII '?' (63 or 0x3F).
> 
> The "ok" will come from the new line.

I noticed that terjeio changed the send function of the `report_init_message` function from the previous `hal.stream.write` to `write`. `"write"` does not send. I don't know if he had another intention by doing this, or if it's purely a typo.

closed 2025-02-12 by hanke-cnc

dresco, 2025-02-12:
I can't see how that's related? Am not able to replicate here with the latest code..

hanke-cnc, 2025-02-12:
> I can't see how that's related? Am not able to replicate here with the latest code..

"`write` does not send a startup message. For my use of canlde, it relies on the startup message to determine whether the connection is successful during the connection. This is only a small issue; I can just switch back to `hal.stream.write`."

dresco, 2025-02-12:
> write does not send a startup message. For my use of canlde, it relies on the startup message to determine whether the connection is successful during the connection. This is only a small issue; I can just switch back to hal.stream.write.

Probably better to get to the root cause. Perhaps @terjeio will have some ideas..

I think the core change was related to [this](https://github.com/grblHAL/STM32F4xx/issues/206) issue.. I have just checked, and for me the startup message **is** being generated correctly when a USB connection is established, and also at board reset/startup.

terjeio, 2025-02-12:
My guess is that Candle does not assert the DTR line as it should on connect. To be fixed in Candle?

hanke-cnc, 2025-02-13:
> My guess is that Candle does not assert the DTR line as it should on connect. To be fixed in Candle?

Strange, the current grbl has become very weird. The following three commands all fail to receive status replies, only "ok" is ever received, but your ioSender works. I don't know what to do.

    qDebug()<<"状态发送";
              //  SerialIf_Write("?");
                 SerialIf_Write(QByteArray(1, (char)0x80));
             //    SerialIf_Write(QByteArray(1, (char)0x05));

hanke-cnc, 2025-02-13:
> My guess is that Candle does not assert the DTR line as it should on connect. To be fixed in Candle?

Oh, that's not right. Receiving a "好的" reply is another timer sending a "$G" message, which means sending "问号" or 0x80 or 0x05 does not receive a reply.

hanke-cnc, 2025-02-13:
> My guess is that Candle does not assert the DTR line as it should on connect. To be fixed in Candle?

Oh no, I found that the issue is with the USB virtual serial port (PA11.PA12). When I use a regular serial port to send "?", I receive a response.

hanke-cnc, 2025-02-13:
> Probably better to get to the root cause. Perhaps [@terjeio](https://github.com/terjeio) will have some ideas..
>
> I think the core change was related to [...]

dresco,In my latest tests, I found that when using a USB virtual serial port, the host computer does not receive a status reply when sending "?", but it can receive replies when sending other commands like "$$". Strangely, using ioSender allows for receiving status replies. Ordinary serial ports function normally.

hanke-cnc, 2025-02-13:
I can only test issues; I cannot debug, locate, or fix code.

I have an H743 and an H723. I believe I can be quite certain that the issue is with the USB virtual serial port code.

terjeio, 2025-02-13:
> Also, why can your ioSender obtain a status reply.

Could be due to you disabling legacy realtime commands with `$39=0` and ioSender [requesting it with a top-bit set character (0x80)](https://github.com/terjeio/ioSender/blob/4ef2873888b5e7bc888e0447696ef1cc4fe3148f/CNC%20Core/CNC%20Core/Grbl.cs#L3017-L3095)?

I added this option to allow printable realtime command characters in string settings such as SSIDs and passwords.

Do you get a response if you connect with a terminal program?

hanke-cnc, 2025-02-13:
> Could be due to you disabling legacy realtime commands with `$39=0` and ioSender [requesting it with a top-bit set character [...]

How can I enable automatic status reporting? Currently, with GRBL, besides setting `$481=100`, it seems like additional commands are needed to activate automatic status updates. I am using the latest STM32H750_GRBL and encountering extremely strange issues—so strange that I can't describe them in words. I have uploaded a video.

https://github.com/user-attachments/assets/6e0a6e60-714c-4f81-8289-d3d8fdbdc7e2

hanke-cnc, 2025-02-13:
> Could be due to you disabling legacy realtime commands with `$39=0` and ioSender [requesting it with a top-bit set character [...]

terjeio，Do you have a PayPal donation link? Please give me one.    You should post your PayPal donation link on the Core homepage so that people can thank you.

terjeio, 2025-02-13:
> How can I enable automatic status reporting? Currently, with GRBL, besides setting $481=100, it seems like additional commands are needed to activate automatic status updates.

No additional commands, but a hard reset (or power cycle)  is required:

[image]

>  I have uploaded a video.

I am not able to parse this. Candle is not working and CNCjs is? And that auto reporting should not output to the USB stream unless DTR is asserted (I'll have to check this)?

> terjeio，Do you have a PayPal donation link?

No, I only accept donation of hardware that I can use for development.

hanke-cnc, 2025-02-13:
> No additional commands, but a hard reset (or power cycle) is required:
>
> [image]
>
>
> I am not able to parse this. Candle is not working and CNCjs is? And that auto reporting should not output to the [...]

Oh, I experimented again. It is unrelated to the automatic status sending (the automatic status sending doesn't seem to work either). I have to first connect to grblhal using cncjs once, then close cncjs and use candle2 to send "?" to receive status replies normally. So I wonder, for the current grblhal, what other commands should I send?

The standard serial ports PA9 and PA10 are functioning normally; there's no need to do this.

terjeio, 2025-02-13:
>> I have uploaded a video.

> I am not able to parse this. Candle is not working and CNCjs is? And that auto reporting should not output to the USB stream unless DTR is asserted (I'll have to check this)?

Testing with a F412 I see that no output is going to the USB port unless a client is connected (detected by DTR asserted) - this is how it should be. So Candle should assert DTR on connect and deassert on disconnect. If not deasserted on disconnect the USB buffer will fill up and that may block output, possibly forever, even if a new client is connected. If this is the underlying issue you may try connecting with ioSender with _Toggle DTR_ enabled - perhaps that will unblock it?

[image]

> Are you saying that you either cannot watch the video I uploaded or don't understand what the video is trying to convey?

I do not understand it. If other senders works the issue is likely with Candle which is written for Arduino based controllers that will be hard reset on connect. Most of the grblHAL supported controllers cannot due to either not having the circuitry for it or that a hard reset will kill the connection if it is native USB (CDC) or network based. More [here](https://github.com/grblHAL/core/wiki/For-sender-developers).

> Yes, I would like to donate you a Nucleo-H723ZG.

Thanks for the offer,but  I think I do not need one since I have a H7xx base BTT Scylla controller I can use for testing.

> I have to first connect to grblhal using cncjs once, then close cncjs and use candle2 to send "?" to receive status replies normally. So I wonder, for the current grblhal, what other commands should I send?

Well, then I am pretty sure Candle does _not_ assert DTR on connect. And CNCjs possibly not deasserting on disconnect... So add code to assert DTR on connect - the serial library should have a function or property for that. Here is [how I toggle control lines](https://github.com/terjeio/ioSender/blob/4ef2873888b5e7bc888e0447696ef1cc4fe3148f/CNC%20Core/CNC%20Core/SerialStream.cs#L124-L143) in ioSender.

> The standard serial ports PA9 and PA10 are functioning normally; there's no need to do this.

Handshake is generally not implemented for UART ports, any output will disappear into the void. For virtual USB ports it will block until the data is taken by a client - that is why grblHAL now uses the (virtual) DTR (Data Terminal Ready) line to determine if a client is connected. This was necessary to avoid the controller hanging e.g. if an UART connected MPG (offline controller) is in use with no USB client connected. Complicated...

dresco, 2025-02-13:
Google suggests the Qt serial port behaviour is different between Windows and Linux, appears that it does not set DTR on Windows, but does elsewhere..

I was able to recreate the issue on my Linux dev box by forcing DTR off after connection, so I think the below Candle2 patch will work for Windows use (untested as I don't have a Windows box here);

```
diff --git a/src/frmmain.cpp b/src/frmmain.cpp
index 1046b0e..7878c90 100644
--- a/src/frmmain.cpp
+++ b/src/frmmain.cpp
@@ -2099,7 +2099,10 @@ void frmMain::on_btnConnect_clicked()
                 ui->txtStatus->setStyleSheet(QString("background-color: palette(button); color: palette(text);"));
 #ifndef WINDOWS
                 SerialIf_Clear();
+#else
+                SerialIf_SetDTR(true);
 #endif
+
                 qDebug() << "Serial OK";
 
                 m_timerRead.start(ReceiveTimerInterval_ms);
diff --git a/src/interface/SerialInterface.cpp b/src/interface/SerialInterface.cpp
index 1d6a03d..7232c51 100644
--- a/src/interface/SerialInterface.cpp
+++ b/src/interface/SerialInterface.cpp
@@ -101,6 +101,11 @@ void SerialIf_Clear()
     m_serialPort.clear();
 }
 
+bool SerialIf_SetDTR(bool state)
+{
+    return m_serialPort.setDataTerminalReady(state);
+}
+
 QString SerialIf_GetError()
 {
     if(m_Interface == IF_SERIAL)
diff --git a/src/interface/SerialInterface.h b/src/interface/SerialInterface.h
index 2693b9e..d7f77e2 100644
--- a/src/interface/SerialInterface.h
+++ b/src/interface/SerialInterface.h
@@ -25,6 +25,7 @@ void SerialIf_Close();
 bool SerialIf_IsOpen();
 
 void SerialIf_Clear();
+bool SerialIf_SetDTR(bool state);
 
 QString SerialIf_GetError();
 
```

hanke-cnc, 2025-02-13:
> Testing with a F412 I see that no output is going to the USB port unless a client is connected (detected by DTR asserted) - this is how it should be. So Candle should assert DTR on connect and [...]

Thank you, it is indeed a DTR problem, you have cured my headache.    😄😄😄😄

hanke-cnc, 2025-02-13:
> Google suggests the Qt serial port behaviour is different between Windows and Linux, appears that it does not set DTR on Windows, but does elsewhere..
>
> I was able to recreate the issue on my Linux [...]

Yes, it is a DTR problem.

dresco, 2025-02-13:
> I think the below Candle2 patch will work for Windows use

@hanke-cnc If you are able to confirm the above patch works for you, I'll try raising a PR to the Candle2 project..

hanke-cnc, 2025-02-13:
> > I think the below Candle2 patch will work for Windows use
> 
> [@hanke-cnc](https://github.com/hanke-cnc) If you are able to confirm the above patch works for you, I'll try raising a PR to the Candle2 project..

Thank you for your kindness, but you don't have to do this. Candle and candle2 are no longer maintained by the authors. Additionally, if candle and candle2 are to be used with grblhal, further changes are needed to accommodate new features. I use candle only because Qt can run on ARM embedded and Windows devices. I will provide a separate copy to those who purchase the grblhal controller board. People should promptly switch to ioSender or cncjs; they are well adapted.

Also, dresco is setting 550mhz for H723 still on your plan list? I always feel it is a bit imperfect, and I cannot do it. As long as this thing is still on your plan list, I hope you do not delete it.