dresco/STM32H7xx #40: SKR 3 / STM32H723 PWM not working on any of the ports

eyadid, 2025-03-21:
Hey. Great work with this project!
I'm having an issue with PWMs on my SKR 3 / STM32H723. None of them work - Spindle outputs (FAN outs) or Servo output (GPIOE-15 / TMR15) are not outputting anything..  **I`m suspecting something is going on with the timers (?) .**

Iv'e tried many configurations using the GRBLHAL framework - none worked. 

My outputs are claimed correctly AFAIK ([AUX IO:2,3,0,1])

Is there any way for me to debug it and help tracing the issue down?
Are there any known issues with PWM setup on this board ? 

```
[VER:1.1f.20250311:]                                                          
[OPT:VNMCDSL,100,1024,3,0]                                                    
[AXS:3:XYZ]                                                                   
[NEWOPT:ENUMS,RT+,NOPROBE,SED,FS,TMC=7,SD]                                    
[FIRMWARE:grblHAL]                                                            
[SIGNALS:RHS]                                                                 
[NVS STORAGE:*FLASH]                                                          
[FREE MEMORY:205K]                                                            
[DRIVER:STM32H723]                                                            
[DRIVER VERSION:250311]                                                       
[BOARD:BTT SKR-3]                                                             
[COMPATIBILITY LEVEL:10]                                                      
[AUX IO:2,3,0,1]                                                              
[PLUGIN:Bootloader Entry v0.02]                                               
[PLUGIN:FS stream v1.00]                                                      
[PLUGIN:FS macro plugin v0.17]                                                
[PLUGIN:Trinamic v0.26]                                                       
[PLUGIN:PWM servo v0.03]                                                      
[KINEMATICS:CoreXY v2.02]                                                     
[PLUGIN:SDCARD v1.23]

```

dresco, 2025-03-21:
Hi, thanks for the report.

> Are there any known issues with PWM setup on this board ?

Not that I was aware of. I have a H743 SKR3 here, so will check it out tomorrow..

eyadid, 2025-03-21:
Iv'e done some more digging and got it to work. 
I`m trying to get PWM running using TIM15.
The fix was modifying these statements to include TIM15:
 
https://github.com/dresco/STM32H7xx/blob/da788a266473a123494fec027c67ced730b16eaa/Src/ioports_analog.c#L142

`        if(pwm_value == aux_out_analog[port].pwm->data.off_value) {
            if(aux_out_analog[port].pwm->data.always_on) {
                *pwm->ccr = aux_out_analog[port].pwm->data.off_value;
                if(pwm->timer == TIM1 || pwm->timer == TIM15)
                    pwm->timer->BDTR |= TIM_BDTR_MOE;
                *pwm->ccr = 0;
            } else {
                if(pwm->timer == TIM1 || pwm->timer == TIM15)
                    pwm->timer->BDTR |= TIM_BDTR_MOE;
                *pwm->ccr = 0;
            }
        } else {
            *pwm->ccr = pwm_value;
            if(pwm->timer == TIM1 || pwm->timer == TIM15)
                pwm->timer->BDTR |= TIM_BDTR_MOE;
        }`

I don't know much about this microcontroller - but I believe that setting the BDTR to TIM_BDTR_MOE enables the timer (?)

Not sure about the repercussions for other boards / functionality but so far my board is doing what it`s suppose to do (give or take)

dresco, 2025-03-22:
Thanks, good catch. It appears timers 1, 8, 15, 16, & 17 use the BDTR register on the H723/H743 parts.

I'll extend that check to cover all of those (not all the timers have outputs defined in pwm.c, but might save tripping over the same thing again if more pinouts are added in the future).

Just want to check something else though. You also mentioned that spindle outputs were not working for you, was that with the PWM Servo plugin also? The regular spindle PWM output on PB0 (EXP1 header) doesn't appear to use that code path, and does work okay for me here with the default SKR3 map..

commit f0be0a50 (2025-06-07): Confgure BDTR register for more timers

Timers 1, 8, 15, 16, & 17 use the BDTR register on the H723/H743 parts.

fixes #40
closed 2025-06-07 by dresco