dresco/STM32H7xx #41: Can you come up with a spindle instructions, as a novice I don't know how to open macros

MCULJ, 2025-04-19:
Compared to an 8-bit GRBL, it's too complicated

hanke-cnc, 2025-04-19:
> 与 8 位 GRBL 相比，它太复杂了

"I don't quite understand what you mean. To turn on the spindle, please use the standard G-code 'M3SXXXX'. If your control board has just been powered on and the spindle speed is zero, then M3 will not actually produce an output (or activate the spindle)."

MCULJ, 2025-04-21:
What I want to know is how the spindle synchronization function is wired and turned on, how should I choose a spindle model, and can I give a specific model

dresco, 2025-04-21:
Hi @MCULJ , if it's not a question specific to the H7 port, you're probably better off asking in the main grblHAL [repo](https://github.com/grblHAL/core). 

As far as I am aware, spindle sync requires either an external encoder or possibly a [stepper spindle](https://github.com/grblHAL/core/discussions/696) (experimental)..