dresco/STM32H7xx #42: Tool Setter Configuration for BTT Scylla Board

theoutfield, 2025-05-19:
Thanks for all the work you did setting these boards up to work with grblHAL it's very appreciated.  I recently purchased a BTT Scylla board to replace a Mach3 board and I'm in the process of compiling and configuring the software.  I don't have any experience doing this so my knowledge isn't great.  I was able to import the project into PlatformIO and compiled/downloaded the firmware to the board.  I was looking through the my_machine.h and btt_scylla_map.h files to see how all the inputs and outputs were configured.  I'm not 100% sure of how the spindle and probe are being enabled but it seems like the configuration is there in the header files and the documentation kind of indicates they are enabled by default.  If not could you give me some direction on this. 

I couldn't find any information regarding the tool setter configuration.  The Scylla board has a tool setter input at PE7 that I would like to use.  Can I get some help in understanding what I need to do to enable that in the firmware.  

Thanks,
Tony

dresco, 2025-05-20:
Hi Tony,

Probe input should get enabled by default unless explicitly disabled. I think the default spindle configuration is PWM controlled, can be overridden at build time if using something else.

Your options for building would either be through the hosted [web builder](https://svn.io-engineering.com:8443/), or locally with PlatformIO or STM32CubeIDE. The web builder should cover common configuration options..

If building locally with PlatformIO, you would typically add additional symbol definitions to the relevant build environment in platformio.ini, this would be the  `btt_scylla_h723_tmc5160_bl128` section for the Scylla board with bootloader support.

The Scylla hardware has a probe input on PE15, and a toolsetter input on PE7. The Scylla board map uses the PE15 input (AUXINPUT1) for probing. Are you using separate probe and tool setter? I **think** that people using both with grblHAL just wire them up together to the same input?

Cheers, Jon.

dresco, 2025-05-20:
Actually, it seems there has been some recent support for a 2nd probe & tool setter [added](https://github.com/grblHAL/core/commit/6948b0e117118de30fdd830f678a221b21e53276) to the core code. That might make things easier?

I've been travelling a lot over the last couple of months, so the H7 port has got a bit behind. Am back home now, so will try to get caught up over the next week or so..

> ...code exists in the Scylla header but I didn't find any reference to where PROBE_ENABLE was set other than the line in the my machine header which said it was enabled by default so I just left it.

That one is set [here](https://github.com/grblHAL/core/blob/070f80cc5c0d095ff63d5980795b5a15c5bb0202/driver_opts.h#L90-L92) in the core code, there will be similar for spindle setup. If you right click on the symbol in VSCode, then "Go to Definition" should point you in the right direction.

For VFD support, you would add something similar to `-D SPINDLE0_ENABLE=SPINDLE_HUANYANG1` to platformio.ini (depending on VFD model).

theoutfield, 2025-05-20:
Thanks again Jon.  I'm just getting into the grblHAL code and wasn't aware of the driver_opts.h file.  What I've read so far led me to believe that the core files shouldn't be edited but I guess that doesn't mean that default values aren't set there.  I appreciate the response and I'll look into the new 2nd probe stuff as well.  

I'm going to hook up some hardware in the next couple of days so I'll figure out what is and what isn't working.  I might have some more questions if you don't mind.

patcarter883, 2025-05-26:
Hi Tony,

I've just recently replaced a Mach3 setup with the BTT Scylla and it seems I had some pretty similar needs as you.

I haven't had a chance to look at and play around with the new core code for a secondary probe setup, I used the probe selector 2 plugin that has worked excellently, it's here https://github.com/grblHAL/Templates/blob/master/my_plugin/Probe_select2/my_plugin.c

The RS485 spindle control was fairly easy to get working, most of the faffing around was on the VFD control side, enabling RS485 and selecting it as the control input etc. 

If your VFD drive is one that is directly supported in the spindle plugin then it is pretty much as simple as  enable RS485 and plugging it in. If it's not a listed VFD you'll be in my position where you'll need to use the MODVFD spindle, and do some researching and math to get the registers and command words to configure the spindle plugin, and get the correct values for the RPM get and set multipliers and dividers.

Feel free to reach out if you need any help. I had great fun setting it up and think the Scylla board is quite good, particularly at the price point, there is just a shortage of information about getting it running compared to other boards.

And many thanks to @dresco to laying the groundwork by having an STM32H7 configuration I could use, love your work good sir.

theoutfield, 2025-05-27:
Hey Pat,

Yeah that would be great to touch base with someone else using the Scylla board.  You can shoot me an email if you want at tony dot fountaine @ gmail.com I finally got my board completely installed and the spindle operating using an analog 10V signal today.  I've got an MK100 VFD which I think is support using RS485 but I haven't tried it.  I was pretty happy to find the board.  I was waiting for the Scienci Labs SLB EXT to be released but I'm glad I found this instead and like you say there is nothing even close to it's price point.  I actually put up a playlist of shoddy videos on YouTube because I noticed there wasn't much info on the board yet [https://www.youtube.com/playlist?list=PLgCmNhrVBcXp4KaI3E3DT8NJ11kGZTyMw](url) .  I'm still waiting for a 24V power supply to finish up the main installation then I'm going to put in a button box for Cycle Start, Pause and E-Stop on the side of my machine.  I'm going to take a look at your like for the probes as well.  

Tony

agri-pat, 2025-05-28:
@dresco Are you happy to take pull requests for your repo. I've updated the Scylla board map file, mostly just comments to identify what the pins are labelled with on the board, as well as a few extra AUX ports assigned to support probe and toolsetter with the probe select plugin.

dresco, 2025-05-28:
> Are you happy to take pull requests for your repo. I've updated the Scylla board map file..

@agri-pat Yes please, thanks!  Am a bit behind vs upstream changes, but trying to get caught up..

closed 2025-06-11 by theoutfield