dresco/STM32H7xx #48: Serial don't work

MrKapim, 2025-06-14:
Hi @dresco , we have 3 serials configured but just one works properly. Is defined USART1, 2, 3 and 6. I'm perform tests in all ports and the results is just the USART3 works. I'm use the keypad stream to send commands and the MPG stream to receive datas. All serials (USART1, 2, 3 and 6) send datas but just USART3 receive datas. We have error in hardware configuration (STM32H743)? Or maybe the interrupt collect have error?

MrKapim, 2025-06-17:
Okay, I Understand you. But why just the IRQ3 works? This is the problem. The IRQ 1 and 2 don't answer properly... And because this, I don't have the other serials working

dresco, 2025-06-17:
Can you provide a failing configuration I can test with?

MrKapim, 2025-06-20:
I'm using the standard configuration. Just insert MPG (stream 0 and enable 2) and Keypad (stream 0 and enable  2 (in my machine.h) and in btt_skr_v3.0_map insert/change #define SERIAL_PORT 1 (to use pins A9 A10). The results is tx OK but rx NOK...

MrKapim, 2025-06-20:
A little video showing the problem...    

https://github.com/user-attachments/assets/e0d2abea-2e68-49c8-a634-a744b64b5a9e

MrKapim, 2025-06-20:
Other detail...  in btt_skr_v3.0_map have a definition for auxoutput5 using the same pin for usart (pin B10). Here I changed to other pin because the TX is all time ON (maybe pull up)...

dresco, 2025-06-23:
Thanks, will have a look this week..

MrKapim, 2025-07-03:
Yes, this is the problem >>> On USART 1, 2, 3 and 6 the real time report works, on USART 2 and 3 Tx and Rx OK but on USART 1 and 6 only Tx OK, Rx NOK). Considering this configuration that you are using (pins A9, A10 + SERIAL_PORT 1 + STREAM 0 + MPG ENABLE 2), it means that you are using USART1 and you should have the same error. Question... Are you sending 0x8B to enter MPG mode?

dresco, 2025-07-03:
> About pin B10 see the printscreen of the file here on GitHub

That serial port definition is not used in the SKR3 board map. If changing the map file, you will need to check the h/w schematic to see whether any other pinouts already defined in serial.c are suitable.

> On USART 1, 2, 3 and 6 the real time report works, on USART 2 and 3 Tx and Rx OK but on USART 1 and 6 only Tx OK, Rx NOK).

Not here, have successfully tested USART1/2/3/6 on multiple boards.

> Question... Are you sending 0x8B to enter MPG mode?

I have done yes, though I usually just test for a response from sending '?' And to clarify, I test with no sender connected (and no automatic reporting), so only get real time reports from the input I send from the terminal.

As I am not able to reproduce what you are seeing, can I suggest you share a repo with all your changes?

MrKapim, 2025-07-03:
Okay. And after you send the 0x8B, you receive "MPG:1"???

MrKapim, 2025-07-03:
Hi @dresco ...   There's something strange going on, but let's get this straight... I downloaded the driver again and adjusted it so that we're the same, on the same page.  
1. Defined this in platformio...

[image]
[image]

2. The sender show me this...

[image]

3. My results is the same, My results is the same before...  no answer or no RX working...

https://github.com/user-attachments/assets/3c2e470f-4b2e-4fc9-9dfb-93f6255e2de5

MrKapim, 2025-07-03:
I've already looked in the logic and found nothing, it should work correctly. The only difference I see is in the H743 hardware, where the USARTs are separated...

[image]

Is there a bug???

dresco, 2025-07-04:
Thanks, have been testing with H743 hardware (Nucleo & SKR3).

There was already a fix committed for serial config when coming from the bootloader [here](https://github.com/dresco/STM32H7xx/commit/dad365f97c66e740197b2a056b11c87d07a7fb10). 

I'll flash the BTT bootloader back onto my SKR3 & test again, in case I've missed something. Give me a day or two..

dresco, 2025-07-04:
Have retested after reinstalling the bootloader, and reflashing the `btt_skr_30_h743_tmc5160_bl128` firmware. Is still working okay for me here.

Only connections to the board are the stepper motors, USB cable, and FTDI cable to PA9/PA10 on the TFT header.
Only changes to platformio.ini are the MPG_STREAM=0 & MPG_ENABLE=2 definitions for that environment.
No other code changes from current master.

Attaching my binary in case you want to compare..
[btt_skr_30_h743_tmc5160_bl128.zip](https://github.com/user-attachments/files/21064840/btt_skr_30_h743_tmc5160_bl128.zip)

MrKapim, 2025-07-14:
Hi @dresco ...  It's finally working!!! We have 3 serials communicating.  But I needed to make some changes... For USART1 (pins 9 and 10), I put a 2k2 resistor in series between TX (pin 9) and my FTDI (RS232 to USB). For USART6, I made a jumper in the electronics (in my case, on the MBB BTT SKR3-EZ chip U12, pins 5 and 6) and changed the USART6 initialization to SWAP (swap the TX and RX pins). Thanks for your effort and patience in helping...

MrKapim, 2025-07-14:
Complementing... For USART6 I am using pins C6 and C7 and the SWAP is fixed in my serial 3 (in serial.c serial2SetBaudRate)