dresco/STM32H7xx #54: how to connect SD card

MatiM72737, 2025-09-26:
hi, i've got problem with SD card connection, how should i connect that to use it with nucleo h723zg i know i need to connect that via breakout but how to make that? (for now nothing worked for me)

dresco, 2025-09-26:
Hi,

I'll verify there hasn't been any regression this weekend, but possibly a signal integrity issue, how did you get your SD card connected? I ended up making a little breakout PCB with series termination etc.

I did the original dev work with a generic board & jumper wires, but I had to slow the clock down - are you using the NUCLEO_SLOW_SDMMC_CLOCK symbol? May also be worth reducing the GPIO drive strengths [here](https://github.com/dresco/STM32H7xx/blob/947ffe0e3b018a492a1f21b6626633ec9cfe2905/Src/stm32h7xx_hal_msp.c#L114-L127).

But anyway, I will re-check this weekend, and make sure there is nothing that's changed since..

MatiM72737, 2025-09-27:
yeah, I'm using NUCLEO_SLOW_SDMMC_CLOCK, just trying with default nucleo_h723_reference_cdc env 
```
[env:nucleo_h723_reference_cdc]
board = nucleo_h723zg
board_build.f_cpu = 480000000L
board_build.ldscript = STM32H723ZGTX_FLASH.ld
build_flags = ${common.build_flags}
              ${usb_h723.build_flags}
              ${sdcard.build_flags}
              ${eth.build_flags}
  # See Inc/my_machine.h for options
  -D BOARD_REFERENCE=
  -D NUCLEO_H723=
  -D IS_NUCLEO_DEVKIT=1
  -D HSE_VALUE=8000000
  -D RTC_ENABLE=1
  -D NUCLEO_SLOW_SDMMC_CLOCK=
lib_deps = ${common_h723.lib_deps}
           ${usb_h723.lib_deps}
           ${sdcard.lib_deps}
           ${eth.lib_deps}
lib_extra_dirs = ${common.lib_extra_dirs}
                 ${usb_h723.lib_extra_dirs}
                 ${sdcard.lib_extra_dirs}
                 ${eth.lib_extra_dirs}
# Alternatively, place the .pio/build/<env name>/firmware.bin on the NODE_H7xxZI drive
upload_protocol = stlink
```
and i'm connecting that with jumper wires ofc, but it's strange for me, because when i try to acces webui later i can see files from sd with $F+ same as before try to acces that ui, i think in theory on error it should disconnect sd, but i'm not sure :)
and what do U mean with reducing drive strenghts here:
```
    GPIO_InitStruct.Pin = GPIO_PIN_8|GPIO_PIN_9|GPIO_PIN_10|GPIO_PIN_11
                          |GPIO_PIN_12;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF12_SDIO1;
    HAL_GPIO_Init(GPIOC, &GPIO_InitStruct);

    GPIO_InitStruct.Pin = GPIO_PIN_2;
    GPIO_InitStruct.Mode = GPIO_MODE_AF_PP;
    GPIO_InitStruct.Pull = GPIO_NOPULL;
    GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_VERY_HIGH;
    GPIO_InitStruct.Alternate = GPIO_AF12_SDIO1;
    HAL_GPIO_Init(GPIOD, &GPIO_InitStruct);

    /* SDMMC1 interrupt Init */
    HAL_NVIC_SetPriority(SDMMC1_IRQn, 0, 0);
    HAL_NVIC_EnableIRQ(SDMMC1_IRQn);
```
, thanks for reply

MatiM72737, 2025-09-28:
now, after making cables shorter (I found SD card socket and soldered it) with 51k ohm pull-ups on data pins and cmd and now totally i can't acces SD and getting error :/ 

[image]
[image]
[image]

(yes, i checked connections, all of them are good)

dresco, 2025-09-28:
I suspect a regression, am seeing checksum differences when uploading/downloading test files through WebDAV.

Leave it with me to check further...

dresco, 2025-09-28:
Appears to be a DMA / cache coherency issue. As a workaround, you can change [this line](https://github.com/dresco/STM32H7xx/blob/947ffe0e3b018a492a1f21b6626633ec9cfe2905/platformio.ini#L50) to 0 to disable the L1 cache.

Regarding the drive strength, you can try changing the two `GPIO_InitStruct.Speed` entries from `GPIO_SPEED_FREQ_VERY_HIGH` to something lower (options are LOW, MEDIUM, HIGH, and VERY_HIGH). I should probably do this anyway if `NUCLEO_SLOW_SDMMC_CLOCK` is defined.

Where in the world are you based? I have spare SD breakouts, and could send one for the cost of shipping, but might not be practical depending on location..

commit 49acf519 (2025-10-02): Ethernet DMA fix

MPU region for LwIP heap was not correctly aligned (base address
of a region should be aligned on its size).

Regions for Rx pool and Tx buffers adjusted accordingly.

Fixes #54